espruino/BangleApps #288: Suggestion: Screenshots for Apps

brainfoolong, 2020-04-13:
For the web based view on https://banglejs.com/apps/, i really miss one thing since the very first beginning: Screenshots. I want to see, how this will look, directly on this page.

In my opinion it could be accomplished very easily:

1. Allow a folder named `screenshots` in an app directory.
2. Filter out all JPG/PNG and if one exist, display a screenshot button in the app row on the website.

If i could help you with that, let me know what i should do. I probably can modify the needed scripts that generate the website.

[image]

paulcockrell, 2020-04-13:
If you find the Mario Clock in the app store, notice its `Read more...` link, click that link to open its README file, in which you see a screen shot of the app. Is that what you are after?

brainfoolong, 2020-04-13:
Oh. Yep, that's what i searched for actually. Didn't knew of the Readme feature.

closed 2020-04-13 by brainfoolong