espruino/BangleApps #294: Back button in sub-menus issue (e.g. App/Widget Settings, Select Clock etc.)

PaulMcGuinness, 2020-04-14:
This is what I did:-

1. switch off
2. Press BTN1 to start
3. Go into setting
4. Go into App/Widget settings 
5. Select "Back"
6. Select "Select clock" in setting menu

Nothing happens.

Same happens if you use "Select clock" then press "Back" - The "Back" button seems to be the issue.

gfwilliams, 2020-04-14:
Are you sure your settings app is up to date?

PaulMcGuinness, 2020-04-14:
All up to date (I'm checking daily at the moment as there are quite a few changes going on)

ps-igel, 2020-04-14:
Just testet this. I can't even enter the App/Widget settings. Web IDE shows

`Execution Interrupted during event processing.
New interpreter error: LOW_MEMORY,MEMORY`

`>process.memory()
={ free: 845, usage: 1255, total: 2100, history: 0,
  gc: 0, gctime: 5.95092773437, blocksize: 16, "stackEndAddress": 536923400, flash_start: 0,
  "flash_binary_end": 483740, "flash_code_start": 1073741824, flash_length: 524288 }
`

I am using settings app v0.13. When I switch back to v0.11 it works again (this one still has issue [#206](https://github.com/espruino/BangleApps/issues/206) though)

gfwilliams, 2020-04-15:
> Web IDE shows "LOW_MEMORY,MEMORY"

Is this completely standard from 'install default apps', or did you add more apps/widgets?

@PaulMcGuinness do you have the standard set of apps, or more apps/widgets installed? I just tried "Install default apps" and did exactly your steps and it works perfectly

PaulMcGuinness, 2020-04-15:
All the apps I have (all updated to latest versions are):-

About
Alarms
App Manager
Astrocalc
Battery Widget
Bluetooth Widget
Bootloader
Compass
GadgetBridge
Firmware Version Widget
GPS Info
GPS Time
Heart Rate Monitor
Launcher
Moon Phase Widget
Morphing Clock
Party Parrot
Pedometer Widget
Settings
Stopwatch

ps-igel, 2020-04-15:
I installed update [0.14](https://github.com/espruino/BangleApps/commit/d54c0c7dea9dcf2d944f1ecd10a64fa7d50984f9) yesterday, it seems to have fixed the app/widgets menu. 

Right know it has:
- Battery widget
- Toucher
- Numerals Clock
- GadgetBridge

Moving back and forth between the submenus works fine.

The clock menu however still causes:

`Execution Interrupted during event processing.
New interpreter error: LOW_MEMORY,MEMORY`

The clocks I have installed:
- Blob Clock
- Numerals
- Mario Clock
- Pipboy
- Word Clock

gfwilliams, 2020-04-15:
Thanks - just updated and fixed this. The clock settings worked like app settings and would run out of memory too easily.

ps-igel, 2020-04-15:
Great - seems to work. However I'm curious if `process.memory()` is actually a good way to debug this. Free memory for the 0.15 update seems to be actually less sometimes. However the menu still works.

Before update:
>New interpreter error: MEMORY
Execution Interrupted during event processing.
>process.memory()
={ free: 839, usage: 1261, total: 2100, history: 0,
  gc: 0, gctime: 6.37817382812, blocksize: 16, "stackEndAddress": 536923400, flash_start: 0,
  "flash_binary_end": 483740, "flash_code_start": 1073741824, flash_length: 524288 }

After update: 
 >process.memory()
={ free: 832, usage: 1268, total: 2100, history: 10,
  gc: 652, gctime: 6.89697265625, blocksize: 16, "stackEndAddress": 536923400, flash_start: 0,
  "flash_binary_end": 483740, "flash_code_start": 1073741824, flash_length: 524288 }

gfwilliams, 2020-04-15:
It is - it gives you a good base figure. The issue was that the code allocated a whole bunch of memory in one go, so even though the base amount of free memory was ok, it must have just allocated more than 800 vars :(

ps-igel, 2020-04-21:
Can this issue be closed then?

PaulMcGuinness, 2020-04-21:
Nope. Just blew new firmware on it, and picked up watch from the desk (Wake on twist) and frozen mid-morph again.

gfwilliams, 2020-04-21:
That's https://github.com/espruino/BangleApps/issues/279

Yes, I think this one is fine

closed 2020-04-21 by gfwilliams