espruino/BangleApps #295: Battery Chart breaks app/widget settings

Didgeridoohan, 2020-04-14:
Testing some of the new apps I noticed that @msdeibel's Battery Chart (v0.05) seems to break App/Widget Settings (tested with Settings v0.13). With Battery Chart installed it's simply not possible to enter that sub menu.

msdeibel, 2020-04-14:
Hmm, Battery Chart doesn't even use Settings. I just updated to settings 0.13 and cannot confirm this. I can enter the submenu for Gadgetbridge and BatterWidget and also make changes in there.

Didgeridoohan, 2020-04-14:
I'll check more. It might be that there are other apps/widgets clashing as well. I tired several times though, and it was reproducible. It also caused my connection to the App Loader to fail while retrieving my apps...

msdeibel, 2020-04-14:
Aww damn. In line 11 of the widget I currently have `var settings = {};`. That could be the culprit, even though I don't experience the issue as I said. I'll fix that at some point tomorrow.

Didgeridoohan, 2020-04-14:
I just found that it only seems to happen if I also have Battery Widget with percent installed.

Purple-Tentacle, 2020-04-14:
I suspect a memory issue.
I have the same problem with my 'Active Pedometer' widget.
There should be messages in the IDE.

Didgeridoohan, 2020-04-14:
I'll check the IDE and see what pops up.

Didgeridoohan, 2020-04-14:
Yup, definitely a memory thing. Seems like I simply had too many widgets and not anything specific to a certain widget.

Purple-Tentacle, 2020-04-14:
Problem is, that widgets are loaded and displayed in the settings app. So they take a lot of memory.

gfwilliams, 2020-04-14:
Is it specifically the app/widget settings menu that breaks? Because maybe we could tweak that to avoid requiring so much memory?

Didgeridoohan, 2020-04-14:
From what I saw I couldn't enter either the Apps/Widget settings or the Select clock sub menus.

gfwilliams, 2020-04-14:
Just pushed some changes that should help with memory usage

Didgeridoohan, 2020-04-14:
Seems to have done it's job just right. No issues now (except if you install way more widgets than could even be displayed, but then things really start to break and you can't even open the launcher).

gfwilliams, 2020-04-15:
I'll close this for now then.

I wonder whether we need something in the bootloader that'll handle the low memory event and display a warning?

closed 2020-04-15 by gfwilliams