espruino/BangleApps #296: Active Pedometer widget takes too much memory

Purple-Tentacle, 2020-04-14:
I need someone to assist me saving memory in my widget.
My dev version can be found here: https://github.com/Purple-Tentacle/BangleAppsDev/tree/master/apps/pedometer

I know that I am using long variable names and a lot of comments, but I am hoping for a minifier to get implemented into the app loader soon.

Where can memory be saved otherwise?

gfwilliams, 2020-04-14:
This sort of thing might be better discussed in the forum so this repo doesn't get clogged up with generic questions, but I took a quick peek, and:

* moving `DEFAULTS` into the function where it's used would help you a little.
* deleting `pedomData` after use would help
* try not to keep the `settings` object around as that could end up being quite large

... that might shave a bit of space off

lixas, 2020-04-14:
variable stepsDisplayLarge is used only once, so maybe there is no point to declare that. I would use it directly in g.drawString function @ 133 line. Also, that would fix bug when step is taken right after midnight- step=1 but in higher font- yesterdays data :)

gfwilliams, 2020-04-23:
Closing this now. Also this kind of thing is probably better done on the forum...

closed 2020-04-23 by gfwilliams