espruino/BangleApps #308: App/Widget settings all loaded into memory

Purple-Tentacle, 2020-04-16:
All of the app/widgets settings are loaded into memory, as soo, as you enter the menu.
For me, memory usage goes from 74% to 86%.

And I can see a point where to many settings are loaded.

Easy workaround:
Do not load and show widgets in the settings app.

gfwilliams, 2020-04-16:
> Do not load and show widgets in the settings app.

Yeah, except I got grief a while back because widgets WEREN'T shown, which is why they got added.

Settings aren't actually loaded since the functions are in flash - so it's just what is required for what is currently on the screen.

Honestly, if the widgets are taking up too much RAM with settings, chances are they won't work well with clocks or other apps that display them.

Purple-Tentacle, 2020-04-16:
> Yeah, except I got grief a while back because widgets WEREN'T shown, which is why they got added.

Haha :)

> Honestly, if the widgets are taking up too much RAM with settings, chances are they won't work well with clocks or other apps that display them.

I just saw the highest memory utilization ever when going into this menu. Even higher as with widgets + clock. That why I figured that could be a problem.
Settings main menu: 62% utilization
App/widget settings: 83%
Clock & widgets: 63%

Where do the 21% come from? seems a lot.

gfwilliams, 2020-04-16:
Do you have the latest settings app? It changed recently to really cut down on used memory

Purple-Tentacle, 2020-04-16:
Yes, 0.15

gfwilliams, 2020-04-16:
Try 0.16 now. I think that should help a lot.

Purple-Tentacle, 2020-04-16:
with 0.16:

Settings main menu: 62% utilization
App/widget settings: 81%
Going back to settings main menu: 82%
Clock & widgets: 63%

gfwilliams, 2020-04-16:
Ahh. This is actually a side-effect of *when* the RAM widget is rendered I think. It only updates every 10 seconds. If you leave it on the menu for >10 seconds it'll probably update to a much more reasonable value.

closed 2020-04-16 by gfwilliams

Purple-Tentacle, 2020-04-16:
Okay, new values with 15 seconds waiting time after going into menu ar back.

Settings main menu: 75%
App/widget settings (3 entries + 'back': 68%
Going back to settings main menu: 76%
Clock & widgets: 64%

So just rounding I think.
Much better :)