espruino/BangleApps #335: BTN4 and BTN5 for moving up/down in settings

ps-igel, 2020-04-21:
The buttons on the watch seem to be a weakpoint (at least to me). They tend to get stuck or get greased up if they don't get pushed regularly. I would like to avoid using them whenever possible.

Having said that: can we make the settings app additionally respond to BTN4 (touch left) for up and BTN5 (touch right) for down? So you would only have to press BTN3 to select/enter/exit a menu item.

brainfart-be, 2020-04-25:
I have the same issue with the buttons getting stuck and/or triggering neighbour button. 

If you want to go full on touch event
```
Bangle.on('touch', function(button){
  // button == 1 : left side of the screen
  // button == 2 : right side of the screen
  // button == 3 : both together
})
```

There is also the swipe event
```
Bangle.on('swipe', dir => {
  // dir == 1 : right to left
})
```

You can take a look at "Toucher" (the launcher)

ps-igel, 2020-04-25:
Yeah, I'm using Toucher as my launcher. But I'm talking about the settings app. It is using E.showMenu which renders the menu and is thus out of my control even if I wanted to modify it myself.

brainfart-be, 2020-04-26:
After playing around a bit (late night work well for me)
I discovered that when `E.showMenu(something);` is called it return a Instance of that menu
While looking at this link http://www.espruino.com/graphical_menu 
I discovered that you can call `.move()`and `.select()`on a menu 

Which mean that if you keep a reference to the current menu you can handle pretty easily the touch event for the settings app.

There is a simple example, I can try to implement it on the settings app if you don't feel like it ;)

```
var currentMenu;

var boolean = false;
var number = 420;

var mainmenu = {
  "" : { "title" : "-- Main Menu --" },
  "Backlight On" : function() { LED1.set(); },
  "Backlight Off" : function() { LED1.reset(); },
  "Submenu" : function() { currentMenu = E.showMenu(submenu); },
  "A Boolean" : {
    value : boolean,
    format : v => v?"On":"Off",
    onchange : v => { boolean=v; }
  },
  "A Number" : {
    value : number,
    min:0,max:100,step:10,
    onchange : v => { number=v; }
  },
  "Exit" : function() { currentMenu = E.showMenu(); },
};

// Submenu
var submenu = {
  "" : {
    "title" : "-- SubMenu --"
  },
  "One" : undefined,
  "Two" : undefined,
  "< Back" : function() { currentMenu = E.showMenu(mainmenu); },
};

// keep a reference to the current menu when you show it
currentMenu = E.showMenu(mainmenu);

// handle the touch
Bangle.on('touch', e => {
 if(!currentMenu) return;
  if(e == 1){
    currentMenu.move(-1);
  }else if(e==2){
    currentMenu.move(1);
  }else{
    currentMenu.select();
  }
});
```

ps-igel, 2020-04-26:
Oh ok, thanks for investigating. It didn't read to documentation careful enough. I will have a look into this, hopefully without running into #206 again.

gfwilliams, 2020-04-27:
You could potentially override `E.showMenu` if you want to do this? Either add your own code and call the existing function, or set `E.showMenu` to https://github.com/espruino/Espruino/blob/master/libs/js/banglejs/E_showMenu.js with your mods in it

gfwilliams, 2020-04-27:
Just to add to this - if you did:

```
E.showMenu = ....
```

In a file called `newmenu.boot.js` then you could happily customise the menu however you see fit, and could even distribute that new menu as an app.

ps-igel, 2020-04-27:
Thanks, I wasn't aware you could overwrite the function that easily. I'll give that a try.

ps-igel, 2020-04-27:
Ok, so I added BTN4 and BTN5 like you suggested and uploaded it as newmenu.boot.js:

```
E.showMenu =
(function(items) { 
  if (Bangle.btnWatches) {
    Bangle.btnWatches.forEach(clearWatch);
    Bangle.btnWatches = undefined;
  }
...
l.draw();
  Bangle.btnWatches = [
    setWatch(function() { l.move(-1); }, BTN1, {repeat:1}),
    setWatch(function() { l.move(-1); }, BTN4, {repeat:1}),
    setWatch(function() { l.move(1); }, BTN3, {repeat:1}),
    setWatch(function() { l.move(1); }, BTN5, {repeat:1}),
    setWatch(function() { l.select(); }, BTN2, {repeat:1})
  ];  
  return l;  
})
```
That works fine, however: what if I want to add touch (BTN4+BTN5) as select()?
If I add it through a `Bangle.on("touch", e=> {if (e==3) l.select();});` it will be recreated for every submenu and I get double/triple touch events.

How can I delete the touch event at the start of the code once it is set?

gfwilliams, 2020-04-27:
http://www.espruino.com/Reference#l_Object_removeListener

ps-igel, 2020-05-04:
Thanks @gfwilliams I got this working using `Bangle.removeAllListeners('touch')`. Closing the issue then.

closed 2020-05-04 by ps-igel