espruino/BangleApps #349: Joystick app

nxdefiant, 2020-04-25:
Hi,

I'm currently working on a simple Joystick App using the accelerometer as stick input, but it does need an extra hid descriptor. If you are interested in adding the app there are basically two options:

1. Extend the HID descriptor in boot0.js with the joystick one
2. Execute the NRF.sendHIDReport() call in the application itself, downside: The application will only work when settings.HID is disabled.

Any preference?

gfwilliams, 2020-04-27:
Hmm. That's a hard one. We could do something like:

```
  if (s.HID) {
    if (s.HID=="joystick") Bangle.HID = E.toUint8Array(atob("..."));
    else Bangle.HID = E.toUint8Array(atob("..."));
    NRF.setServices({}, {uart:true, hid:Bangle.HID});
  }
```

Then we could make the type of HID device optional - and apps and then easily check if it's set to the right one?

Otherwise I guess we could store the actual descriptor in settings if it's non-standard, which could be more flexible...

nxdefiant, 2020-04-27:
After some more testing I would actually prefer option #1, since the Bluetooth hosts does not always seem to react well to the descriptor change: Sometimes I had to to restart the host for the new service to get recognized.

Also extending the descriptor should not be a problem to the two exiting profiles. Obviously it hardly scales with every new Bluetooth app..The next person wants to add a mouse descriptor...

> Otherwise I guess we could store the actual descriptor in settings if it's non-standard, which could be more flexible...

Flexible yes, but I think harder to manage. We would need some kind of interface for the user to select which services he wants out of (keyboard, multimedia, joystick) and build the descriptor accordingly.

gfwilliams, 2020-04-27:
Ok then, sounds like a plan. Do you have the descriptor that you'd like me to add?

nxdefiant, 2020-04-27:
Sure, this one:

```
0x05, 0x01,       // Usage Page (Generic Desktop)
0x09, 0x04,       // Usage (Joystick)
0xA1, 0x01,       // Collection (Application)
	0x09, 0x01,       // Usage (Pointer)
	0xA1, 0x00,       // Collection (Physical)
		// Buttons
		0x05, 0x09,       // Usage Page (Buttons)
		0x19, 0x01,       // Usage Minimum (1)
		0x29, 0x05,       // Usage Maximum (5)
		0x15, 0x00,       // Logical Minimum (0)
		0x25, 0x01,       // Logical Maximum (1)
		0x95, 0x05,       // Report Count (5)
		0x75, 0x01,       // Report Size (1)
		0x81, 0x02,       // Input (Data, Variable, Absolute)

		// padding bits
		0x95, 0x03,       // Report Count (3)
		0x75, 0x01,       // Report Size (1)
		0x81, 0x03,       // Input (Constant)

		// Stick
		0x05, 0x01,       // Usage Page (Generic Desktop)
		0x09, 0x30,       // Usage (X)
		0x09, 0x31,       // Usage (Y)
		0x15, 0x81,       // Logical Minimum (-127)
		0x25, 0x7f,       // Logical Maximum (127)
		0x75, 0x08,       // Report Size (8)
		0x95, 0x02,       // Report Count (2)
		0x81, 0x02,       // Input (Data, Variable, Absolute)
	0xC0,              // End Collection (Physical)
0xC0              // End Collection (Application)
```

So the new descriptor becomes:
`Bangle.HID = E.toUint8Array(atob("BQEJBqEBhQIFBxngKecVACUBdQGVCIEClQF1CIEBlQV1AQUIGQEpBZEClQF1A5EBlQZ1CBUAJXMFBxkAKXOBAAkFFQAm/wB1CJUCsQLABQwJAaEBhQEVACUBdQGVAQm1gQIJtoECCbeBAgm4gQIJzYECCeKBAgnpgQIJ6oECwAUBCQShAYUDCQGhAAUJGQEpBRUAJQGVBXUBgQKVA3UBgQMFAQkwCTEVgSV/dQiVAoECwMA="));
`

gfwilliams, 2020-04-28:
Not sure if the base64 is right... the length of that is 179, but the array above it is only 52?

I'll stick the actual array contents in

nxdefiant, 2020-04-28:
Thats the complete base64 descriptor including the other two keyboard and multimedia.

gfwilliams, 2020-04-28:
Ahh, wow. I was thinking we'd just make it change to be *just* a Joystick?

I just pushed some changes in https://github.com/espruino/BangleApps/commit/517c35a37ddaed8716ea19785d77dc5c4aaa1432 that should make this work nicely for you (I hope!)

nxdefiant, 2020-04-29:
One minor change please. if it is "just a joystick" please exclude the report id in the hid descriptor. The text in hid_info.txt already has it excluded, but not the base64 in boot0.js, it should become:
> 'BQEJBKEBCQGhAAUJGQEpBRUAJQGVBXUBgQKVA3UBgQMFAQkwCTEVgSV/dQiVAoECwMA='

Keyboard works fine with the unpatched Linux/bluez, but you should be aware that on Linux I had to restart bluetoothd everytime I switched the HID setting. On Android I had to disable and enable bluetooth. This might produce a lot "it does not work" bug reports. This is the full test matrix:

```
App	\ 	Setting:	Keyboard + Media	Keyboard	Joystick
Keyboard->Linux			ok			ok		-
Media->Linux			ok			-		-
Joystick->Linux			-			-		ok
Keyboard->Linux(unpatched)	error(1)		ok		-
Joystick->Linux(unpatched)	-			-		error(2)
Media->Linux(unpatched)		error(1)			-		-
Keyboard->Android		ok			ok		-
Media->Android			ok			-		-

error(1) = expected by the bluez report id problem/bug
error(2) Needs removal of report id in joystick descriptor to work
```

gfwilliams, 2020-04-29:
> The text in hid_info.txt already has it excluded, but not the base64 in boot0.js

That's odd - I generated the base64 right from it - but sure, I'll stick the new one in.

Thanks for doing that testing. Honestly I find myself restarting bluetoothd pretty often on Linux anyway, so I don't think this is really anything new :)

nxdefiant, 2020-04-29:
Looks good, I'm closing this ticket, thank you.

closed 2020-04-29 by nxdefiant