espruino/BangleApps #354: AES encryption

paulcockrell, 2020-04-26:
I was messing about with some bluetooth smart devices, and they use aes 128 encryption. Does the Bangle have an implementation of AES?

lixas, 2020-04-26:
>`typeof AES.encrypt`
Uncaught ReferenceError: "AES" is not defined
 at line 1 col 11
It seems bangle does not have AES support by default
`require("crypto")` also does not help. https://www.espruino.com/modules/crypto.js here on only SHA1 implementation

paulcockrell, 2020-04-26:
Yes I saw this old thread regarding the Puck, and assumed they had the same features  http://forum.espruino.com/conversations/319871/

gfwilliams, 2020-04-27:
Bangle.js doesn't have the implementation I'm afraid. Adding it would be as simple as adding it to the `BANGLEJS.py` and rebuilding but it's likely other features (tensorflow?) would have to be removed to make space.

> crypto.js

Yes, the AES implementation is part of the firmware and is written in C. The JS file there is just a filler for devices that don't even have the crypto lib but want SHA1

Any chance you can ask these kinds of questions in the forum in future? Otherwise it gets hard to figure out which issues actually need stuff doing on them

closed 2020-04-27 by gfwilliams