espruino/BangleApps #364: App sorting order in launcher

jl-cs, 2020-04-29:
Is there a way to sort the installed apps in which order the appear when in the launcher?
It also might be possible to write an app for that so the sortorder could be adjusted directly on the watch. Would someone be interested in something like that, besides me?

Thanks in advance for your feedback.

gfwilliams, 2020-04-30:
Right now there's no way to specify it via UI, but as you noted there is a `sortorder` field. I think that'd be a really neat addition.

I guess the only issue is upgrading apps would likely change the sort order. To work around that I guess it could be implemented in the App Loader?

For on-watch, if you create a file called `foobar.settings.js` then the Settings menu will pick it up (under 'App/Widget Settings') so that'd be a really easy way to add the option for people if they wanted it.

jl-cs, 2020-04-30:
Thank you for your feedback!

Could you elaborate what you mean by implementing it in the app loader? What would that change in terms of the upgrading issue?

Also, could you please point me to the part of the documentation where I can read up on the usage of the `foobar.settings.js` file? Is this a kind of plug-and-play mechanism to add a submenu in the settings-app for the sorting functionality I suggested?

Again, thanks for your feedback and also all the work you do on Espruino. You're amazing!

gfwilliams, 2020-04-30:
Hi! 

> Could you elaborate what you mean by implementing it in the app loader?

I mean have a menu in https://banglejs.com/apps/ 

That's mean that the app loader could be aware of a change in sort order for apps, and would know to change the `sortorder` field on any new `app.info` files it uploaded.

> where I can read up on the usage of the foobar.settings.js

https://github.com/espruino/BangleApps#adding-configuration-to-the-settings-menu

> Is this a kind of plug-and-play mechanism to add a submenu in the settings-app for the sorting functionality I suggested?

Yep, that's right :)

jl-cs, 2020-04-30:
Thanks for the quick response.

> I mean have a menu in https://banglejs.com/apps/
> 
> That's mean that the app loader could be aware of a change in sort order for apps, and would know to change the `sortorder` field on any new `app.info` files it uploaded.

Now I get what you mean, I've confused the app loader with the app launcher.

> > Is this a kind of plug-and-play mechanism to add a submenu in the settings-app for the sorting functionality I suggested?
> 
> Yep, that's right :)

Ok then this will be my starting point. Thanks again.

One last question: would a pull request to the settings app be a good idea to publish the `foobar.settings.js`-version when done or what would you propose in order to make it publicly available for the bangle.js community?

gfwilliams, 2020-04-30:
I'd just add another 'app'. Just an entry in `apps.json` mentioning your settings.js file should do it - but if you have any troubles let me know and I can sort that bit out

jl-cs, 2020-05-05:
Thinking about the right place for this I think it might fit better into the existing "App Manager" App [https://github.com/espruino/BangleApps/tree/master/apps/files](url)
I think I'll add a submenu there so the "Settings" app won't be clogged even more. My watch is already running near RAM max. capacity when using the "Settings" app.

jl-cs, 2020-05-10:
I finally found the time, so please see [this pull request](https://github.com/espruino/BangleApps/pull/401)

closed 2020-05-11 by jl-cs