espruino/BangleApps #365: Bug: Morphing Clock: 27:58 instead of 07:58

myxor, 2020-04-30:
This morning i was able to see this on the Morphing Clock:

The watch was lying on the desk over night and it showed me **27:58** instead of the **07:58** which would be correct.
After holding BTN3 it switched back to the correct time.

See attached picture:

[image]

gfwilliams, 2020-04-30:
Interesting - thanks! Are you sure the clock firmware was up to date? There were a few changes recently that should have helped with stuff like this.

myxor, 2020-04-30:
Ah sorry, forgot to write: i am on 2v05.439 firmware and v0.04 of the Morphing Clock.

myxor, 2020-05-02:
The same happened again this night with the 2v05.27 firmware build.

[image]

gfwilliams, 2020-05-05:
Ok, thanks. Got this fixed now :)

commit 7d8d1a8f (2020-05-05): Add "ram" keyword to allow 2v06 Espruino builds to cache function that needs to be fast
      Fix issue where first digit could get stuck going from "2x:xx" to " x:xx" (fix #365)

also remove accidental commit
closed 2020-05-05 by gfwilliams