espruino/BangleApps #376: Single digits date isn't 0 padded

nikmartin, 2020-05-04:
https://github.com/espruino/BangleApps/blob/51bf1d4ae4a297513e58fdd49f3a940a8c09ca1b/apps/aclock/clock-analog.js#L87

Not sure if there's a bug in my implementation, or a bug in ternary ifs in Espruino, but this line doesn't work in the analog clock app.

gfwilliams, 2020-05-04:
It's just to do with operator precedence. What you're doing is basically:

```
```

closed 2020-05-04 by gfwilliams

nikmartin, 2020-05-04:
Ah, for some reason I have this:
`const dateString = ((currentDate.getDate() < 10) ? '0' : '') + currentDate.getDate().toString();`
in my local repo, but it never got committed. That's what I get for developing on two different computers! The extra parens set the proper precedence.

commit 0d9eeab6 (2020-05-04): aclock: Fix broken date padding (fix #376)