espruino/BangleApps #380: [Feature request] Settings app: remember main menu location when in submenu

Bzly, 2020-05-05:
Currently, if one scrolls through the Settings app and then enters a submenu (i.e. Locale), the 'back' button in the submenu takes you to the top of the main menu, rather than your last position (expected and nicer behaviour, imo).

gfwilliams, 2020-05-05:
Yeah - it would be better, but is nontrivisal. We've had some memory usage issues so would have to do some slight hackery to make sure we passed the menu position around without keeping the entire old menu in RAM.

gfwilliams, 2021-02-02:
Closing this for now - I've tried to add a bunch more stuff into submenus so the main settings menu is nowhere near as long as it was and I think this is less annoying now

closed 2021-02-02 by gfwilliams