espruino/BangleApps #381: Bug: Morphing Clock

jl-cs, 2020-05-05:
Don't know if anyone else has the same problem, but when using the Morphing Clock and the LCD turns of for some time, the clock doesn't update itself when the LCD is turned on again. Event the seconds counter has stopped and it only returns to normal if I switch to menu and back by BTN2 or hold BTN3 to reset the clock.
Any ideas why this might be happening?

Bzly, 2020-05-05:
Which version of Morphing Clock do you have? The App Loader should tell you on the "My Apps" screen. 

From my cursory glance this looks similar to #279 - reportedly fixed in v0.04+

jl-cs, 2020-05-05:
I'm on v0.04 of Morphing Clock and also latest(?) firmware according to this post: [http://forum.espruino.com/conversations/346029/](url)

gfwilliams, 2020-05-05:
Could you try turning `Debug Info` to `Show`? And maybe also install the `RAM` widget.

It's possible you have enough stuff the clock is running out of memory, or maybe some other widget is causing an error which is stopping the clock from working.

jl-cs, 2020-05-05:
Yes, `Debug Info` is set to `Show`.
I'm also using the RAM Widget , but while daily usage to only other things running are the "Battery Level Widget (with percentage)", "Gadgetbridge" (with active bluetooth connection to my smartphone) and "Battery Chart". Then, when activating the watch it shows the RAM at 54% while clock + these widgets are in use. I don't know if that is already too much. Also, when the clocks hang, the RAM Widget doesn't show a strongly different amout of used RAM, at max. 2% or 3% more then the "normal"(?) 54%.

jl-cs, 2020-05-07:
Is there a way to log the running code even further? Despite using the latest versions of all apps from here [https://espruino.github.io/BangleApps](https://espruino.github.io/BangleApps) the problem persists and even got worse, since the freezes happen very often.

gfwilliams, 2020-05-07:
Yes, sure. The best way to debug is to connect with the Web IDE: https://www.espruino.com/ide/

Then hopefully if you can get the problem to occur, the IDE will display the full error message and stack trace that caused the issue.

jl-cs, 2020-05-08:
@gfwilliams I've did have connected the watch with the IDE when the error occured, but sadly I've neither got any output for the error in the Web IDE, nor in the console window of the Web IDE. Do you have any other suggestions what I could try? Maybe a complete (factory) reset of the watch? Or can you think of anything else I could try beforehand?

jl-cs, 2020-05-10:
@gfwilliams I've found the time to check the problem again and it turned out it was a RAM issue after all. But I still have problems understanding why exactly, since no errors were shown.
I've had an additional Widget running, which I forgot. It was the "Chrono Widget" and after I removed it, the freezes got rarer, but still occured. Since removing that Widget already freed around 5% of RAM I removed another, namely the "Battery Chart" Widget. Now I always have a RAM usage of below 45%, only sometimes when a Gadgetbride notification comes in, it raises to around 48-50% and since then no freezes have occured anymore.
That's a good think naturally, but what puzzles me is the question, why the Watch froze the LCD in the first place. Before the Widget-removing it used just around 10% more in daily use, but then the Clock constantly froze, though it had around 40% of RAM free. That brings me to my questions:
- Is this expected behaviour? If yes, how much RAM can I safely use? Why do I seem to be unable to use more than around 50% RAM without the Clocks freezing?
- If this is not expected, is my watch broken? (or it's RAM?)

Any help would be much appreciated.

gfwilliams, 2020-05-11:
Ok, that's interesting - so you think the battery chart causes the crashes? I guess it might actually be related to that memory leak I just found.

> If yes, how much RAM can I safely use? 

The issue is whether an app/widget tries to use a chunk in one procedure call. If it does that call might fail. It really depends what is installed. I'm very surprised you don't get stack traces though.

> Why do I seem to be unable to use more than around 50% RAM without the Clocks freezing?

It might be related to one of the widgets that's on there. The memory widget only updates every 20 seconds (iirc?) so it may not reflect the current memory usage at all.

> If this is not expected, is my watch broken? (or it's RAM?)

No, I think the watch is fine!

jl-cs, 2020-05-11:
Thanks again for the feedback. I'm not sure if it actually was just the "Battery Chart" widget and (maybe) the "Chrono Widget" (the one I've deleted first and it got better) or even a combination of all these widgets that were active that tried to allocate memory. If I feel brave enough I might try to flash the updated firmware with the memory leak fix and see if the problem resurfaces when I reinstall the widgets.

> I'm very surprised you don't get stack traces though.

That also surprised me.

> It might be related to one of the widgets that's on there. The memory widget only updates every 20 seconds (iirc?) so it may not reflect the current memory usage at all.

Interesting, good to know.

> No, I think the watch is fine!

That's good to hear. :-)

msdeibel, 2020-05-11:
Thanks @jl-cs for pointing to the issue with the BatteryChart widget. I'll double check my code. There may be an issue with how the `setInterval` is used. Also I might remove the rendering of the widget icon as it ccurrently provides no additional value besides showing that the widget is installed and I trust the users to remember they did so.

jl-cs, 2020-05-11:
@msdeibel I didn't mean to say there were acutal issues I could point to. Only that after i deleted the BatteryChart widget, the real issues got better.
What I want to say is though, that I really liked the BatteryChart widget and I'm looking forward to the next update. Thanks for working on that!

msdeibel, 2020-05-11:
Just released BatteryChart 0.10 on https://msdeibel.github.io/BangleApps/index.html (PR for the master is in) if it reduces the issues and leads to pinpointing the root cause - good. If it fixes the root cause, even better ;)

gfwilliams, 2020-05-11:
You could try using Storage.write(file, JSON.stringify(data)) - basically the same thing, as writeJSON was a utility function

jl-cs, 2020-05-12:
> You could try using Storage.write(file, JSON.stringify(data)) - basically the same thing, as writeJSON was a utility function

That's nice to know, I think I'll try that for the app sorting then.

jl-cs, 2020-05-12:
> You could try using Storage.write(file, JSON.stringify(data)) - basically the same thing, as writeJSON was a utility function

Please see https://github.com/espruino/BangleApps/pull/413

gfwilliams, 2020-05-15:
Does this seem fixed now then?

jl-cs, 2020-05-15:
Yes, for the time being it's working. Though I haven't checked the complete old setup again. But I'll close this issue for now and reopen it if the problems are coming back.
Thanks again for all the help!

closed 2020-05-15 by jl-cs