espruino/BangleApps #382: [Feature request] Launcher app: allow scrolling through the ends of the list

Bzly, 2020-05-05:
Currently, if you have more than a few apps installed it is a slight pain to get to the bottom ones of the list - you just have to go through the entire thing. 

It would be nicer if - for example when at the top of the list - you can press BTN1 and it cycles you round to the bottom. 

This and the discussed sortorder solution in #364 go quite nicely together, and should make the launcher UX a bit better.

closed 2020-05-05 by gfwilliams

gfwilliams, 2020-05-05:
Just added :)

Bzly, 2020-05-05:
Thank you! 

When do changes to the development App Loader get mirrored to the release version (in general, I mean - is it timed or 'as and when')?

gfwilliams, 2020-05-05:
It's more 'as and when'. If I think something big went in then I might delay it a bit to see if anyone complains :)

You can load the new stuff from https://espruino.github.io/BangleApps/ right now though

commit 87cf91d8 (2020-05-05): Allow scrolling to wrap around (fix #382)