espruino/BangleApps #386: [Bug] mclock does not play nicely with 'no launcher' message

Bzly, 2020-05-06:
[image]

Bug introduced in v0.04 - https://github.com/espruino/BangleApps/commit/bf5bf91967eb37d703533ce775f57aa8b31696f8 - as far as I can tell. 

Clock digits get cleared by the no launcher message, then on interval if a digit would change, it is drawn over the top. 

Perhaps non-trivial to solve without getting the entire thing to redraw (bad). Low priority because who walks around rocking no launcher?

gfwilliams, 2020-05-06:
Thanks - that's interesting. I guess most other clocks would have issues too, but might overwrite the whole screen. I guess I could change it so Bangle.js just displays that message for 2 seconds then reboots?

Bzly, 2020-05-07:
Yes, you're quite right, it will and does effect all other clocks that do not entirely re-draw. I don't quite know how I missed that...

gfwilliams, 2022-02-03:
Fixed now I think

closed 2022-02-03 by gfwilliams