espruino/BangleApps #389: Morph clock does not update first digit

avanc, 2020-05-07:
I leave my watch on the bedside table at night and grab it in the morning, it does not update the first digit (2, as go went to bed ad 23:00) anymore. The other digits update (morph) correctly:
[image]

It was 6:49 at that moment.

avanc, 2020-05-07:
I just realised that there is also a glitch in the Bluetooth icon. It shows the grayed out icon correctly, and part of the blue (active) Bluetooth icon. But I'm not sure if this already happened yesterday after I disconnected the watch from charging.

Didgeridoohan, 2020-05-07:
Wasn't this fixed a couple of days ago? Ref:
https://github.com/espruino/BangleApps/issues/365
https://github.com/espruino/BangleApps/commit/7d8d1a8f6f34d135ab9d13fa5438cdac5f8fce10

Have you updated to Morph Clock v0.05?

avanc, 2020-05-07:
Ah, I missed that update. Sorry.

closed 2020-05-07 by avanc