espruino/BangleApps #393: [Feature Request]: QR Code: wifi credentials

v1nc, 2020-05-07:
Storing wifi credentials as QR code on the device would be very useful. Unfortunately the used library does not support this. 

Is it possible to use another library?

jl-cs, 2020-05-08:
You would first need to create this on your device and then send them to an app on your watch I think. The bangle.js doesn't have a Wifi modules afaik.

v1nc, 2020-05-08:
> You would first need to create this on your device and then send them to an app on your watch I think. The bangle.js doesn't have a Wifi module afaik.

Please check the existing QR code app. The QR code is generated in the App Loader and then gets saved on your device. You can also do this for wifi credentials, you just need a library that supports this :smiley:

jl-cs, 2020-05-08:
Ah cool, thanks for the hint, hadn't seen that yet.

gfwilliams, 2020-05-11:
Sounds like a good idea. If you found a library we could include and did a PR I'm happy to change it.

However looking at it the library is https://github.com/davidshimjs/qrcodejs/blob/master/qrcode.js which seems to use this QR code generator: https://kazuhikoarase.github.io/qrcode-generator/js/demo/

AFAIK you can just create some text like https://github.com/zxing/zxing/wiki/Barcode-Contents#wi-fi-network-config-android-ios-11 and shove it in to the existing library?

gfwilliams, 2020-05-11:
Actually just tested and pasting `WIFI:T:WPA;S:mynetwork;P:mypass;;` into the existing QR Code generator works great, so it'd be trivial for you to use the existing lib and add 2 extra fields for wifi name and password:

https://github.com/evgeni/qifi/blob/gh-pages/index.html#L198

v1nc, 2020-05-11:
> Actually just tested and pasting `WIFI:T:WPA;S:mynetwork;P:mypass;;` into the existing QR Code generator works great, so it'd be trivial for you to use the existing lib and add 2 extra fields for wifi name and password:
> 
> https://github.com/evgeni/qifi/blob/gh-pages/index.html#L198
Thanks for your research, I will be happy to make a PR soon.

closed 2020-05-11 by v1nc