espruino/BangleApps #397: Infinite alarm retrigger

massimilianogalanti, 2020-05-10:
Hi, for unknown reason my watch keeps triggering alarm (08:00) over and over...
This prevents any usage of the watch and also persists across forced reboots.

gfwilliams, 2020-05-11:
Hi - have you ever updated your Bangle's firmware? https://www.espruino.com/Bangle.js#firmware-updates.

I'm pretty sure you're hitting this issue: "Settings won't change / Welcome screen keeps appearing"

https://www.espruino.com/Troubleshooting+Bangle.js#settings-won-t-change-welcome-screen-keeps-appearing

In rare cases it was possible for the on-watch storage to get corrupted such that a file couldn't be written with different info - it looks like it has happened here. Following the instructions above should fix it.

nxdefiant, 2020-05-26:
Confirmed. I currently can not exit the alarm. No matter if I select "ok" or "sleep", the alarm just reappears. Unfortunately I currently have bluetooth disabled. Also the long press btn1+2 reset does not exit the loop. The firmware was up to date as last weekend.

nxdefiant, 2020-05-26:
After wildly pressing some buttons I managed to set make connectable=true in settings in between the alarm loop. I can see now two files named "alarm.json" with the same content. After removing one of them the alarm stopped.

gfwilliams, 2020-05-26:
Hi - glad you got it sorted! As I'd said in my last post though, this is fixed if you update your Espruino firmware - not just the >1 filename, but you can also you can now *always* connect to Bangle.js regardless of BLE connection state or what the JS side of things is doing by using https://www.espruino.com/Bangle.js#resetting-without-loading-any-code

So I'd definitely suggest doing an update (even if your watch reports itself as 2v05) to avoid the alarm issue happening again.

closed 2020-05-26 by gfwilliams

nxdefiant, 2020-05-26:
Looks like I can not update from 2v05 to 2v05. Everytime I tried I get a "dfu file error"

gfwilliams, 2020-05-26:
And you downloaded the file from https://www.espruino.com/Download ? Maybe you could try downloading it again in case it was corrupt.