espruino/BangleApps #399: Experimental Hebrew font

yarons, 2020-05-10:
I modified Dennis8 bitmap to support Hebrew, before I go any further, I want to make sure I'm doing the right thing.
[image]

Thanks.

gfwilliams, 2020-05-11:
Wow, that's great - thanks!

A hack to make this work would be to add the font code into the following in a file called `hebrew.boot.js`:

```
g._setFont = g.setFont;
g.setFont = function(f,n) {
  if (f=="6x8") {
   // probably just replace font/widths with the base64 in-place
   this.setFontCustom(font, 32, widths, 8);
  } else this._setFont(f,n);
}
```

The only gotcha is that Dennis8 is 8x8 and the built-in font is 6x8, so some apps may be expecting the font to be a slightly different width.

yarons, 2020-05-11:
[image]
Is this better?

gfwilliams, 2020-05-12:
Looks great! Thinking about it, it's probably best to put the code I posted above in the locale file if using a locale that needs it - so apps that don't use locale won't load it in, but if they use the locale module it'll add it.

So in order to get something like https://github.com/espruino/BangleApps/pull/248 in, I guess we need to specify a codepage. For the font you posted, what codepage is it?

Probably the best thing to do is to add a `codepage` field to the `locale.js` for `he_IL`.

Then `locale.html` can detect this, add the font, and do translation.

To do the actual translation to the codepage, I'm not sure if anything is built in, but maybe the most sane thing to do is to actually write out the mapping. So...

```
var codepages = {
 "ISO8859-8" : { // maybe?
   bitmap : "....",
   widths : "....", 
   mapping : 
    " !"#$%&'()*+,-./"+
    "0123456789...."+
    "..."
};
```

Then `locale.html` could basically just run through the `trans` text like:

```
  result="";
  for (var i=0;i<text.length;i++) {
    var mapped = codepages[codepage].mapping.indexOf(text[i]);
    if (mapped>=0) result+=String.fromCharCode(mapped);
  }
```

gfwilliams, 2020-05-15:
Hi - I'm happy to do all the relevant changes to make this work, but what codepage is this? It doesn't look like ISO8859-8 to me (which would have that char starting at 0xE0):

http://czyborra.com/charsets/iso8859.html

yarons, 2020-05-15:
I'll change this accordingly, sorry.

gfwilliams, 2020-06-01:
Did you get around to this? I've just added https://github.com/espruino/BangleApps/commit/d9442f2fd077eba216abda3ccd41a8e766e1c3aa to fix some other stuff, so if we had the correct font then it should be pretty easy to get this in now

yarons, 2020-06-02:
I need a couple of more days, thank you for checking back with me, I'll try to work on it pretty soon.

closed 2021-02-02 by gfwilliams

barakplasma, 2021-12-04:
Any update on this @yarons ? Should I try to pick this up? I'm interested in adding hebrew support to improve my new hebrew calendar app for PR #996

yarons, 2021-12-04:
Hey @barakplasma It's a copy and paste task with some painting app, I did the initial with some pixel editor online, now if you want this fixed all you have to do is to change their order accordingly and it should be good enough.

I can help with some further explanation but since I'm not usually using a mouse it's very hard for me to do.

gfwilliams, 2021-12-06:
I should add that there's the font converter here: http://www.espruino.com/Font+Converter

And you can see in that you can select a `Range` of characters. It should be possible to change the such that there was another option that renders a completely different codepage out? That might be more flexible

barakplasma, 2021-12-06:
I tried using the font converter, but since hebrew starts from charCode 1488 to 1514, the converter isn't working (usually it works with range up to 255, which does work in ISO-8859-8 , but not in JS). I tried to fix the converter in https://jsbin.com/wawuzoxixu/edit?html,output but I haven't figured it out yet. (I tried using <link href="https://fonts.googleapis.com/css2?family=Cousine&display=swap" rel="stylesheet"> as the input)

gfwilliams, 2021-12-06:
Yes, the idea would be that instead of:

```
var img = drawCh(String.fromCharCode(ch));
```

you do:

```
var img = drawCh(String.fromCharCode(FONTMAP[ch]));
```

And `FONTMAP` is a 256 entry array which contains what character codes go where - possibly you could use the same format as suggested for `mapping` in https://github.com/espruino/BangleApps/issues/399#issuecomment-627169001

barakplasma, 2022-01-27:
I'm stuck; I think I'm going to abandon my PR ( https://github.com/barakplasma/BangleApps/pull/2/files ) for now.
I was able to generate a codepage with a mapping. However, I think the mapping is backwards (right to left), and I am having a hard time moving forward. Let's leave this closed for now.
```js
const codePages={
"ISO8859-8": {
    name: "ISO8859-8",
    bitmap: undefined,
    widths: undefined,
    // eslint-disable-next-line no-irregular-whitespace
    mapping: " ¢£¤¥¦§¨©×«¬­®¯°±²³´µ¶·¸¹÷»¼½¾‗אבגדהוזחטיךכלםמןנסעףפץצקרשת‎‏"
  }
}
```

gfwilliams, 2022-01-27:
Ok, thanks - sorry it's not going. I wonder if maybe because hebrew is read right-left there's something strange going on with the character orderings?

yarons, 2022-01-27:
Yeah, unicode solved that by individual character directionality but it's pretty difficult to implement, both Hebrew and Arabic has numerous browser encoding configurations exactly for that reason, until today there are many GNU utilities that are translated sdrawkcab (backwards 😄).