espruino/BangleApps #403: gpsrec option to change distance units

mcranium, 2020-05-11:
My suggestion is to add a new setting to the app, where one can change the unit of the distance which is displayed when you plot your track via the gpsrec app. This setting could be on the same level as "Time Period" and should at least include the option to change from miles to kilometers. 
I would have tried it myself, but I am completely new to javascript and I did not even find the "ml" string was in the source code. Thank you for your help!

gfwilliams, 2020-05-11:
You mean what is displayed when you view a track?

The distance actually comes from the current locale, so if you install a `Language` app for your region it should set the distance units up correctly :)

mcranium, 2020-05-11:
Exactly. I didn't know that the units were managed in the language packs. I now created my own language pack with English in combination with metric units and it works perfectly fine :)

closed 2020-05-11 by mcranium

gfwilliams, 2020-05-11:
Cool, thanks. I guess that might be an interesting option to add to the `Language` page - 'override units'/etc.