espruino/BangleApps #406: Image Background Clock: 88:88:88

jjok, 2020-05-11:
Hi. I've just updated to the latest version of Image Background Clock and it seems like, each time it draws the time, it's drawing over the top of the old time, without clearing a space. So each character eventually ends up as an `8`. I'm pretty sure it wasn't doing this in version 0.02, which I previously had installed, but I can't see where in the code the problem was introduced. Thanks.

nebbishhacker, 2020-05-12:
I just ran into the same issue on my watch.

As far as I can tell the `imgclock.face.bg` file got corrupted - `require("Storage").read("imgclock.face.bg")` returned an empty string, and I had to run `require("Storage").erase("imgclock.face.bg")` twice in the IDE before it disappeared from storage.

Possibly related to https://github.com/espruino/Espruino/issues/1801?

nebbishhacker, 2020-05-12:
And it happened again! Once again, I was able to fix it by erasing `imgclock.face.bg`. It's like there's a zero-size copy of the file hiding the real file:

```
>let s = require("Storage");
=function () { [native code] }
>s.list();
=[
  ".boot0",
  ".bootcde",
  ".trishig",
  "launch.app.js",
  "setting.app.js",  ...
  "imgclock.info",
  "imgclock.app.js",
  "imgclock.face.img",
  "setting.json",
  "imgclock.face.bg"
 ]
>s.read("imgclock.face.bg")
=""
>s.erase("imgclock.face.bg")
=undefined
>s.list()
=[
  ".boot0",
  ".bootcde",
  ".trishig",
  "launch.app.js",
  "setting.app.js",  ...
  "imgclock.info",
  "imgclock.app.js",
  "imgclock.face.img",
  "setting.json",
  "imgclock.face.bg"
 ]
>s.read("imgclock.face.bg")
="\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7" ... "\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7\xD7"
>s.erase("imgclock.face.bg")
=undefined
>s.list()
=[
  ".boot0",
  ".bootcde",
  ".trishig",
  "launch.app.js",
  "setting.app.js",  ...
  "imgclock.face.json",
  "imgclock.info",
  "imgclock.app.js",
  "imgclock.face.img",
  "setting.json"
 ]
```

gfwilliams, 2020-05-12:
It might be worth updating the firmware just in case. I'm not really sure how a zero size file could happen unless there was some sort of filesystem issue

nebbishhacker, 2020-05-26:
Upgrading to 2v05.73 seems to have done the trick - it's been two weeks and I haven't encountered the issue again.

gfwilliams, 2020-05-27:
Great!

closed 2020-05-27 by gfwilliams

jjok, 2020-06-15:
I've just upgraded to 2v06 and it's now working. 👍