espruino/BangleApps #409: [Feature request] Playlist/random clock on LCD turning on

msdeibel, 2020-05-11:
There are already a few very nice watch faces but we can only select one. It would be awesome if the watch could show a different watch face every time the LCD is turned on. Either always in the same order or at random.

Probably adding a setting in/below `Select clock` that has 
- Selected only
- Top to bottom
- Random
as options

gfwilliams, 2020-05-11:
That sounds cool. I guess there are two ways to do this:

* A 'clock' that just randomly picks a new clock and loads it (but this wouldn't do a new clock face when the watch turned on - just when you used BTN3)
* A widget that responds to `Bangle.on('lcdPower'` and automatically loads a clock with `load()`. The only gotcha is there'd be a bit more of a delay when you wanted to turn the LCD on

msdeibel, 2020-05-12:
That would mean checking all the `.info.js` files, selecting the ones with `"type":"clock"`, picking a random one and the `load("name.app.js")`. Right?

gfwilliams, 2020-05-12:
Yes, that's right. There's already some code if you search the repo for `"clock"` I reckon

msdeibel, 2020-05-12:
So a first version exists in PR #415.

No this has two issues:
1. It kills any app that is currently running when the screen turns on. Is there a possibility to check if the app that is currently running is a "clock" type?
2. To hide the "double" loading of clock apps when the LCD turns on I tried to load the new clock when the screen turns off. This however reactivates the screen. Can a background load be done without switching the LCD on?

gfwilliams, 2020-05-13:
> Is there a possibility to check if the app that is currently running is a "clock" type?

Ahh, yes, that's annoying. Right now there's no straightforward way - I've been meaning to make `load(filename)` set some variable that can be checked. By itself that doesn't 100% help and this has come up before though, so I'd be fine with doing a PR that sets `global.CLOCK=true` for every clock face?

> Can a background load be done without switching the LCD on?

I'm afraid not - while if booting fresh you could temporarily replace `g.flip()`, the `reset/load` actually resets the LCD each time, which means powering it on.

msdeibel, 2020-05-13:
I had the widget run this morning and saw that it's no good. The problem with the load being executed based on the screen turning on is, that it breaks Gadgetbridge notifications. The notification starts to pull down and the load operation kills it.
An alternative would be to have the widget change the "selected clock" setting. This would mean less frequent changes, but is less invasive. Can a widget change these basic settings? If yes, how?

gfwilliams, 2020-05-13:
Yes, I think that'd be tricky. What about the first option i'd suggested with the app?

> Can a widget change these basic settings? If yes, how?

Yes - best to look at what the settings app does for 'change default clock'... It's pretty straightforward
Yes, it's

msdeibel, 2020-05-15:
I added PR #424 to make the widget less invasive. It changes the setting with the same result that a new watch face is loaded on long press BTN 3.

gfwilliams, 2022-02-03:
I think this is fixed with https://banglejs.com/apps/#rndmclk

closed 2022-02-03 by gfwilliams