espruino/BangleApps #417: [New app] Espruino controller

gfwilliams, 2020-05-14:
Just a pre-set list of Espruino commands that can be selected from a menu and maybe a terminal showing the response.

Personally I have Espruinos around the house/garden and it's handy to be able to send a command like `waterTomatoes()`, set the time or similar.

jjok, 2020-05-18:
Would the app have to know which device to send the message to? Would it already know about all your devices?

gfwilliams, 2020-05-18:
I figured for now it'd just scan and show you a list of the devices it found, then give you a list of commands it could send...

But I guess it'd make sense to have the ability to specify 'send X to device Y' without having to manually connect first

jjok, 2020-05-18:
Maybe a configuration like:

| Name | Device | Command |
| ----- | ----- | ----- |
| Water Tomatoes | 12:34:56:78 | waterTomatoes() |

Then, in the app, you just see the names of the commands, and the app knows which device to connect to and what command to send.

gfwilliams, 2020-05-18:
Yes - maybe if 'device' was left blank it'd scan and ask.

commit c449a258 (2020-06-24): Added Espruino control (fix #417)
closed 2020-06-24 by gfwilliams