espruino/BangleApps #420: Upload failed, TypeError: Espruino.transform is not a function

cmF3, 2020-05-14:
After updating to **espruino_2v05.82_banglejs.zip** uploading any app to the Bangle.js via the AppLoader fails with the error message ```Upload failed, TypeError: Espruino.transform is not a function```. This makes the Bangle.js effectively unusable.

Steps to reproduce:
* Flash **espruino_2v05.82_banglejs.zip**
* Afterwards, Connect via the AppLoader
* Choose About -> Install Default Apps
* This will fails with "Unknown Error"
* The Bangle.js now contains 0 Apps
* Attempt to upload any App from the Library
* Observe error message noted above

Didgeridoohan, 2020-05-14:
It's not 2v05.82 specific. I'm seeing the same on 2v05.73, although I haven't installed Default Apps. I just wanted to update the bootloader to v0.17.

johanbove, 2020-05-14:
@gfwilliams I am getting the same error as OP.
"Removing all apps" works, but then the process fails with the "Espruino.transform" error.

johanbove, 2020-05-14:
Checking _espruinotools.js_ and it seems the function "`Espruino.transform()`" is added all the way at the end of the file. This seems to make it miss being included in the final Espruino object. The function declaration looks out of place; could something have gone wrong with the compilation?

johanbove, 2020-05-14:
This could be related to the fixes done for https://github.com/espruino/BangleApps/issues/416

gfwilliams, 2020-05-15:
Damn, sorry - that'll teach me for pushing 'minor' changes at night. I'll get it sorted now

gfwilliams, 2020-05-15:
Just fixed. Please can you try now?

closed 2020-05-15 by gfwilliams

cmF3, 2020-05-15:
I have just tested uploading apps on the dev version of BangleApps and the issue persists.

I did not perform any firmware update or anything else.

gfwilliams, 2020-05-15:
What about https://banglejs.com/apps/ ? Sometimes GitHub tends to cache things for a while.

gfwilliams, 2020-05-15:
Just to add - you shouldn't need a firmware update or anything - it was purely an app loader issue

cmF3, 2020-05-15:
"Install Default Apps" finished without error and the Bangle works again.

Is there any functional difference between the app loader on github.io and on banglejs.com?

gfwilliams, 2020-05-15:
Phew, that's great - thanks!

> Is there any functional difference between the app loader on github.io and on banglejs.com?

The one on banglejs.com can sort by newest/recently updated, but that's it.

Supposedly the github.io one is the newest, but because I can control banglejs.com and I disable caching, if I've pushed some code change right out to both at the same time, chances are it'll hit banglejs.com first - but only by a few minutes

johanbove, 2020-05-15:
Confirming with a quick test in the console of https://banglejs.com/apps/ that "Espruino.transform()" is back. Thank you @gfwilliams

commit 9178e1b6 (2020-05-15): Fix Upload failed, TypeError: Espruino.transform is not a function (fix #420)