espruino/BangleApps #425: [Gadget Bridge] Find Lost Phone functionality not working

cmF3, 2020-05-15:
Gadget Bridge offers the ability to find a lost device by pressing a button in the interface.
Usually this is implemented so that the Device starts vibrating when you press that button and does so until you press 'Found it!' 

When coupled to a Bangle.js, Gadget Bridge offers that button but it does nothing, i.e. the Bangle.js stays silent.

Might this be related to the 'missing speaker' issue?

gfwilliams, 2020-05-18:
Interesting - I don't think it's implemented. There's functionality to go the other way (find your phone) but I don't remember seeing the find your device thing implemented.

The `find` message here: http://www.espruino.com/Gadgetbridge

Should be handled at https://github.com/espruino/BangleApps/blob/master/apps/gbridge/widget.js#L149, but it isn't

Jasper-Ben, 2020-07-07:
it does work, however the Bangle.js has to be set to programmable, which is a security risk.

https://github.com/espruino/BangleApps/issues/78 is the related issue for the security side of things.

gfwilliams, 2020-07-07:
> however the Bangle.js has to be set to programmable

To be fair it has to be set to programmable for Gadgetbridge to work *at all*...

Actually looks like I forgot to update this issue when I fixed it last week (https://github.com/espruino/BangleApps/commit/77de51dac4cd27a3fc526f404134126d18d62870#diff-c3fd0472286fb7f28e25673f7a100eb2) so closing this now

closed 2020-07-07 by gfwilliams

Jasper-Ben, 2020-07-07:
Connecting to the gadgetbridge and the "lost my phone" functionality actually works without having programmable set on (the other way around doesn't). As I didn't know there was work on this issue since, I assumed that this might be the cause :)

enaon, 2020-09-15:
Gatdetbridge sends the command, but sends the stop right after the start buzzing. 
This comes in espruino after a button press on the phone. It works ok if one presses the found button and tries again. 
GB({"t":"find","n":true})
GB({"t":"find","n":false})