espruino/BangleApps #426: [Gadget Bridge] Battery level display missing

cmF3, 2020-05-15:
Gadget Bridge allows to see the battery level of a connected device.

This is missing for the Bangle.js

nebbishhacker, 2020-05-17:
If you have the Gadgetbridge widget installed on the watch it will report the current battery level back to your phone. It only does this when the widget starts up, though, which seems like a bit of an oversight. You can force it to report the current status by holding button 3 to refresh the watch face.

nebbishhacker, 2020-05-17:
I created a pull request which might help: https://github.com/espruino/BangleApps/pull/429

cmF3, 2020-05-18:
I have the Widget installed, but apparently never received any battery updates because the battery level is not displayed on the phone.

I upgraded to v0.11 and now it works! Thank you.

Didgeridoohan, 2020-05-27:
Shouldn't this be closed now that battery is properly reported to GB?

gfwilliams, 2020-05-27:
Yep, thanks!  Didn't read the second sentence of the last post :)

closed 2020-05-27 by gfwilliams

Moini, 2023-03-23:
This doesn't work reliably for me currently.

Moini, 2023-03-25:
Unfortunately, it's hard to reproduce... I was able to make the watch vibrate from the phone, but the battery status would not show up. I didn't even know one can tap the icon, so - yes, the first!

I think it just stopped reporting it, because the line is completely horizontal in the battery graph for about 3.5 days - and I did connect the watch within those days. 

I had issues with Bluetooth connection on one of those days, but I don't remember the time sequence anymore. The watch wouldn't connect to the phone for about one day, which only got resolved by connecting to the desktop browser, restarting the watch, and restarting the phone.

(connecting to desktop also proved difficult, because my bluetooth manager constantly wants to autoconnect, and it doesn't seem to work when the watch is already connected - even though I have autoconnect disabled, now that I noticed it's a problem. If anyone else has this issue: 
1. Chromium does work perfectly, at least on Linux Mint 21.1, despite docs warning about it.
2. Make the adjustments outlined for Chrome.
3. In the Bluetooth manager 'remove' the watch from the list
4. To optimize chances of connection, open Bluetooth settings on the watch, even if all BT options (BT LE, Programmable) are active, and open the 'Make connectable' screen. )

gfwilliams, 2023-03-27:
That is a really strange one... Do you have any custom apps installed? I guess maybe something could be interfering with the data that the 'Android Integration' app is trying to send to Gadgetbridge

>  because my bluetooth manager constantly wants to autoconnect

This could potentially be the cause of a lot of issues for you - is it possible that if the watch became disconnected from the phone, your PC could have been trying to connect.

I use Mint here as my main dev machine, so I know this can work nicely. Maybe you could go into the Bluetooth device manager, find the Bangle and click `remove device` for it? That way it really shouldn't autoconnect - and you just use Chrome to connect

Moini, 2023-03-27:
>  Do you have any custom apps installed?

'Custom' as in, not from the official App website? No. But lots from there.

> click remove device for it

Yes, that's what I need to do every single time I try to connect to the desktop. The autoconnect extension for blueman-manager is deactivated, but it still does that as soon as the bangle is available for connection. Seems a blueman issue, though, not yours.

> is it possible that if the watch became disconnected from the phone, your PC could have been trying to connect

Could be possible, at least during the day. But that would not explain the gaps in data during the night, when my computer is turned off. I need to watch out for any patterns...