espruino/BangleApps #427: Disable UART if blerepl is off?

LinkTed, 2020-05-16:
https://github.com/espruino/BangleApps/blob/fcd5a06e21db51101bfaf3cc5c99af19800ecf40/apps/boot/boot0.js#L13-L16

Is there any reason why the BLE UART service is not disabled if the watch is not programmable?  I thought that the REPL is being transferred through BLE UART.  

According to the documentation the BLE UART service can be disabled with: 
```js
NRF.setServices({}, {uart:false});
```

gfwilliams, 2020-05-18:
I think the reasoning is just that:

* Someone might actually want the UART available to handle stuff themselves 
* iOS (and some newer Androids iirc) cache services, so if they change it can be very confusing for the user

LinkTed, 2020-05-19:
Okay I did not know that.

closed 2020-05-19 by LinkTed