espruino/BangleApps #433: Alarms don't work

m0n5t3r, 2020-05-20:
Looks like the alarm works maybe once or twice, then in subsequent days they don't trigger any more... until I open the settings menu (I thought maybe I was sleeping too hard, but in the last 3 days I woke up before the alarm time, and there was no alarm...)

Where should I look in order to debug this further?

gfwilliams, 2020-05-20:
Hi - have you ever updated the firmware on your Bangle.js? It's possible it is related to a bug in the original firmware which could occasionally stop a file being changed.

To debug though, I'd look at `alarm.json` (you can view it using the Storage menu in the IDE). Hopefully you should be able to see it changing. There's:

* `hr` - the hour/minutes it should fire on
* `last` - the number of the last day the alarm fired on to avoid duplicates
* `on` - whether the alarm is enabled

I guess it's possible that the specific clock face you have is doing something `clearInterval()/etc` which may stop the alarm from being kept scheduled?

m0n5t3r, 2020-05-20:
Last I checked, it's the latest stable version (2v05); I don't see any `clearInterval` in https://github.com/espruino/BangleApps/tree/master/apps/sclock

The alarm isn't lost, btw, it shows up in alarms and gets triggered when I open the ~settings menu~ launcher (and I just looked in the IDE, alarm.json is there).

Everything else is up to date (I updated the bootloader now, it was at 0.16)

Going to try connecting with the IDE before the alarm is supposed to trigger tomorrow morning, just in case something gets printed.

gfwilliams, 2020-05-20:
You could also try running `eval(require("Storage").read("alarm.boot.js"))` if the alarm doesn't fire on time, and see if that does anything?

m0n5t3r, 2020-05-20:
only "undefined", which I guess is expected (looking at the code, the function doesn't return anything)

gfwilliams, 2020-05-20:
Yes - but when the alarm is due to fire it *should* show the alarm on the watch?

m0n5t3r, 2020-05-20:
ah, ok, I missed the last part of your message; if I understand correctly, unless there is an active alarm the app resets the whole thing every 100ms, triggering the bootloader? that seems a bit ... excessive, wouldn't a `setInterval` be more ... power efficient?

gfwilliams, 2020-05-20:
Not sure I understand? https://github.com/espruino/BangleApps/blob/master/apps/alarm/boot.js is what runs at startup. It uses `setTimeout` to schedule loading of the alarm when it's time for it to go off.

https://github.com/espruino/BangleApps/blob/master/apps/alarm/alarm.js is the actual alarm - it should *only* be called when there's an alarm, so if there is no alarm then yes, it just reloads the main app after 100ms

m0n5t3r, 2020-05-21:
just tested, `eval(require("Storage").read("alarm.boot.js"))` makes the alarm show up

gfwilliams, 2020-05-21:
Thanks - that's interesting then. It looks like maybe it wasn't being called in the first place when the clock face started?

Either that or somehow the timeout is getting removed.

If the alarm hasn't fired but it should have you could try typing `global["\xff"].timers` to dump the currently running timers - I guess it's possible the alarm is in there but the timeout was wrong?

m0n5t3r, 2020-05-21:
it seems to vary:

```
>global["\xff"].timers
=[
  undefined,
  { time: 150253952, interval: 314572800,
    callback: function () { ... }
   },
  { time: 9120576, interval: 15728640,
    callback: function () { ... }
   },
  { time: 21090272, interval: 31457280,
    callback: function () { ... }
   },
  { time: 55383712, interval: 62914560,
    callback: function (undefined) { ... }
   }
 ]
>global["\xff"].timers
=[
  undefined,
  { time: 110995328, interval: 314572800,
    callback: function () { ... }
   },
  { time: 1319232, interval: 15728640,
    callback: function () { ... }
   },
  { time: 13288928, interval: 31457280,
    callback: function () { ... }
   }
 ]
>global["\xff"].timers
=[
  undefined,
  { time: 100090016, interval: 314572800,
    callback: function () { ... }
   },
  { time: 6142560, interval: 15728640,
    callback: function () { ... }
   },
  { time: 2383616, interval: 31457280,
    callback: function () { ... }
   }
 ]
>global["\xff"].timers
=[
  undefined,
  { time: 89520448, interval: 314572800,
    callback: function () { ... }
   },
  { time: 11301632, interval: 15728640,
    callback: function () { ... }
   },
  { time: 23271328, interval: 31457280,
    callback: function () { ... }
   }
 ]
```

gfwilliams, 2020-05-21:
Thanks - so the fact that they all have `interval` points to the fact that they were made with `setInterval` and not `setTimeout` - so I guess the alarm timeout never ends up in there :(

m0n5t3r, 2020-05-21:
when I need to procrastinate I'll take a stab at fixing / rewriting the thing; this whole "reload the app 10 times a second" sounds awfully wrong to me, unless I'm missing something big (like setInterval callbacks eating too much memory or something like that)

m0n5t3r, 2020-05-26:
Tracing the logic, there's no 100ms loop, indeed; it looks like the app's bootloader schedules a load of `alarm.js` at the time of the first alarm it finds, only if it finds an alarm that didn't run during the current day; after the alarm runs, `.last` will be the current day, so a subsequent `load()` from `alarm.js` will skip that code completely, unless something runs `load()` or similar after midnight.

At least that's what I think, going to test this hypothesis tomorrow and report back.

m0n5t3r, 2020-05-26:
I think I was right:

```
global["\xff"].timers
=[
  undefined,
  { time: 156678944, interval: 314572800,
    callback: function () { ... }
   },
  { time: 156801216, interval: 629145600,
    callback: function () { ... }
   },
  { time: 15581408, interval: 15728640,
    callback: function () { ... }
   },
  { time: 26460192, interval: 31457280,
    callback: function () { ... }
   }
 ]
>
```

starting and exiting the launcher seems to restart the clock twice, and then there is a timer defined without an interval (so presumably the alarm `setTimeout` - although, if those are supposed to be microseconds it will be about 16 minutes late)

``` ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v05 (c) 2019 G.Williams
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v05 (c) 2019 G.Williams
{"t":"status","bat":100}
>global["\xff"].timers
=[
  undefined,
  { time: 21968567007,
    callback: function () { ... }
   },
  { time: 296999648, interval: 314572800,
    callback: function () { ... }
   },
  { time: 611697760, interval: 629145600,
    callback: function () { ... }
   },
  { time: 14354592, interval: 15728640,
    callback: function () { ... }
   }
 ]```

m0n5t3r, 2020-05-27:
it actually triggered on time, so those aren't microseconds allright...

closed 2020-05-27 by gfwilliams