espruino/BangleApps #449: Speedo app not working

ps-igel, 2020-05-26:
I just realized the Speedo app is not working.

It just sits there with "Waiting for GPS and 0 satellites found". However other GPS apps work.

I did an explicit `Bangle.setGPSPower(1)` and it started working, which would be consistent with the [Bangle.GPS documentation](https://www.espruino.com/Reference#l_Bangle_GPS).

> To get this event you must turn the GPS on with Bangle.setGPSPower(1).

Just makes we wonder, if it never actually worked? I'm pretty sure I tested it in the past.

gfwilliams, 2020-05-26:
It should work... setGPSPower is the first line?

https://github.com/espruino/BangleApps/blob/master/apps/speedo/speedo.js#L1

ps-igel, 2020-05-26:
Ok, I fiddled around. Seems like a widget is interfering with it.

If I move `setGPSPower(1)` directly in front of the `Bangle.on("GPS"...)` after the `Bandle.loadWidgets/drawWidgets` calls then it works fine.

Or alternatively commenting out the load/drawWidgets altogether is works too.

Running widgets right now are Ram, Bluetooth, TinyBattery and GadgetBridge (but a modified version). Will test later if the latter on could be the issue.

gfwilliams, 2020-05-26:
Just looked at this, and if you had 'GPS Recorder' that could have caused the issue, even though it wouldn't show in the widget bar. Just moved setGPSPower to the end now so at least that should fix it.

ps-igel, 2020-05-27:
Ok thanks. Can confirm it is working now.

closed 2020-05-27 by ps-igel