espruino/BangleApps #456: Pretokenisation in IDE issue

paulcockrell, 2020-05-27:
Same issue as reported in #416, but seen in the IDE in Chrome. It doesn't like 'get' methods when pretokenisation is enabled.

The Chrome IDE is reporting its using 2v05.1

gfwilliams, 2020-05-27:
Which IDE? The chrome web app, or espruino.com/ide? and what version does it report? Just under the 'Espruino IDE' text at startup

paulcockrell, 2020-05-27:
Ah sorry espruino.com/ide. It reports 2v05.1

gfwilliams, 2020-05-27:
It should be reporting something like: `Web App, v0.73.7`

Do you have an example of code that has the issue, because I'm pretty sure this got fixed a while back.

paulcockrell, 2020-05-27:
Sorry, I think this is my mistake, it is fine loading example code in to emulator with pretokenisation on, but my watch must be running old firmware where it fails when loading code in to the watch.

My watch is on 2v05.1 where as the emulator is 2v05.82.

Sorry!

```js
const myObj = {
  get thing() {
    return "hello, thing";
  }
};
```

closed 2020-05-27 by paulcockrell