espruino/BangleApps #457: Time Traveler's Chronometer freezes

Didgeridoohan, 2020-05-27:
@hopkira I absolutely love the Time Traveler's Chronometer! Thank you 

Found an issue though...

Whenever Gadgetbridge, or any other app for that matter, connects to the Bangle while the screen is off, and thus turning it on, the whole watch locks up (not when disconnecting though). The watch face gets stuck halfway to being drawn and a BTN1+BTN2 restart is needed to get things working again.

It doesn't matter what colour is selected or if widgets are enabled or not.

Disabling decorations stops it from happening, but they are so pretty so of course you would want those enabled.

Was gonna see if I could catch some errors in the web IDE, but since you can't connect two devices/apps at the same time I wasn't sure on how to go about that. I'm sure there's some command I could run to simulate a Bluetooth connection event, but I don't know what that could be.

At the moment I'm running the Bangle on firmware 2v05.102.

Didgeridoohan, 2020-05-29:
@gfwilliams Is there any command or code I could run in the web IDE to try to replicate this and maybe catch some error output?

gfwilliams, 2020-05-29:
You could try sending `NRF.emit('connected')` - or just a `g.flip()` will turn the screen on.

Didgeridoohan, 2020-05-29:
Issue is that just turning the screen on doesn't cause the lockup. It only seems to happens if connecting to Bluetooth while the screen is off.

NRF.emit('connected') didn't do anything (I guess since it's already connected), and g.flip() just turns on the screen.

hopkira, 2020-06-01:
Hmmm odd, yes I've managed to replicate this; the watch has been running fine for more than a week, but trying to install my latest app had precisely this effect.  Not at all sure why it would get half way through drawing the decoration then hang, there's nothing unusual about that part of the code - but that's precisely the behaviour I see too.

gfwilliams, 2020-06-01:
Can you try turning 'pretokenise' off in the App Loader settings and see if that fixes it?

hopkira, 2020-06-01:
I'm on the latest firmware and have updated to the latest apps and Settings v0.19 but don't see that option in the App Loader or Settings.  So, sorry for being dim! - how do I turn 'pretokenise' off?

nxdefiant, 2020-06-01:
Did you check in https://banglejs.com/apps/ the "About" Tab?

Didgeridoohan, 2020-06-02:
Yes, it's under the "About" tab in the App Loader. But, installing the clock with pretokenise off makes no difference to the issue at hand.

gfwilliams, 2020-06-02:
I just tried to connect with the IDE when off and I don't see this.

Is it just when connecting with the App Loader? I could actually kind of imagine that - the app loader sends 'Ctrl-C' to break out of anything currently running when it connects. Since the redraw takes a while it probably breaks out of that halfway through.

You can trigger it with `Bangle.emit("lcdPower",true)` and then bashing Ctrl-C repeatedly - but that doesn't really break it. Next time the screen powers off and back on it's fine.

Didgeridoohan, 2020-06-02:
@gfwilliams Yes, I can't reproduce the problem when connecting from the web IDE, but connecting from the App Loader, Gadgetbridge or any Bluetooth connection does trigger it.

gfwilliams, 2020-06-02:
But you say you need BTN1+BTN2 to get out? Just holding BTN3 doesn't fix it?

Didgeridoohan, 2020-06-02:
Affirmative. The watch completely locks up. Doesn't react to any button presses, long press BTN3 does nothing, screen doesn't power off... A reset of the watch gets it up and running again.

gfwilliams, 2020-06-02:
Ok, I just figured it out I think. It goes into debug mode - I bet if you look at the debug console in the App Loader it'll show `debug>` somewhere.

I just filed an issue for this on https://github.com/espruino/Espruino/issues/1842

I'm struggling to reproduce it here but it might be that just sending two `\x03` instead of one will fix it.

Didgeridoohan, 2020-06-02:
Checking the console when connecting to the App Loader I get this at the point when it locks up.
```
<BLE> Received "In debug mode: Expec"
<BLE> Received "ted a simple ID, typ"
<BLE> Received "e 'help' for more in"
<BLE> Received "fo.\r\n"
TOAST[error] Device connection failed, undefined
```

gfwilliams, 2020-06-02:
Ok, great - that's it then. Not specifically a Chronometer issue other than that it takes a while to draw

Didgeridoohan, 2020-06-13:
@gfwilliams The https://github.com/espruino/Espruino/commit/46796ea7adfba9242af52cf711708f5d107e0b7a commit did make it possible to exit the frozen Chronometer watch that comes from Bluetooth connecting when the screen is off. Yay.

But what is it that causes the screen to freeze in the first place? If I don't catch it happening the screen never powers off, and battery depletes quite quickly with the screen always on...

gfwilliams, 2020-06-22:
The issue is that if you send Ctrl-C while Espruino is busy executing code and it doesn't finish in a second or so, Espruino enters debug mode so you can see what is broken.

Ctrl-C is sent by the app loader when it connects to clear the input line, and that's what causes the issue.

I'll need to change the App Loader I think, to ensure that it exits debug mode if it gets entered. I'll close this bug when I do that

thyttan, 2022-11-28:
Candidate for being closed - leave a comment if you object!

Rationale:
If the change to the app loader regarding exiting the debug mode is already made I think this issue can be closed.

gfwilliams, 2022-11-30:
Thanks! Yes, I'm pretty sure the app loader is now smart enough to exit debug more

closed 2022-11-30 by gfwilliams