espruino/BangleApps #460: Update app error message in dev app loader

ps-igel, 2020-05-28:
I recently having issues updating apps (using the development app loader at github):

Current case: updating the compass app. After updating begins Chrome displays following error message:

[image]

The app then disappears from "My Apps" but can be found in the Library tab again. However, the files are still on the Bangle.

[image]

I can add the compass through the Libary tab again and it works. Never seen that behaviour before, console shows:

```
puck.js:362 <BLE> Sending "\u0003\u0010reset();\n"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "\u0010E.showMessage('"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "Uploading\\ncompa"
puck.js:362 <BLE> Received "\u001b[?7l\r\n ____        "
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "ss...')\n"
```

I attached the complete log.

[20200528_update_apps_log.txt](https://github.com/espruino/BangleApps/files/4694114/20200528_update_apps_log.txt)

gfwilliams, 2020-05-28:
Thanks for the log - that's really handy. So can you give me the *exact* steps? You go to the dev app loader and click 'connect', then update just the one app? And does it happen all the time? And what device is it on? A phone, or PC?

When the 'compass update fails' appears, is it after the update has finished? eg. is anything else being written to the dev console?

I just tried and I can't reproduce here. I guess it could be related to Gadgetbridge, but even with that installed I don't see issues.

ps-igel, 2020-05-28:
> So can you give me the exact steps?
1) Go to dev app loader and connect
2) Go to my apps for updates, click update app arrow at compass app

> And does it happen all the time?
I'm not sure, but I tend do say yes. It happend at least 3 times now and I think that 3 times were the bootloader, the alarm app and then the compass app

>And what device is it on?
Normal laptop with Windows 10 and Chrome 83.0.4103.61 64bit

>When the 'compass update fails' appears, is it after the update has finished?
It all happens quite fast, but yes I think it is when the progress bar is at the end.

>eg. is anything else being written to the dev console?
I didn't see anything. The log attached above is the complete log from updating including the error and then installing again, nothing removed.

Could it be, that that message is a somehow garbled version of the "Hold BTN3 to reload" message?

gfwilliams, 2020-05-28:
The message it's getting is a line from the text that gets sent when bangle.js resets - so it seems like it's calling reset but then not managing to wait for the response...

ps-igel, 2020-06-02:
It still keeps happening, today with updating the alarm clock to v0.09. 

Chrome shows the error message and did not finish the upload. The watch still shows "Uploading alarm..."

Complete log:
[espruino.github.io-1591090145256.log](https://github.com/espruino/BangleApps/files/4716203/espruino.github.io-1591090145256.log)

Last time I could get it to update eventually, but not today. Resetting and rebooting didn't work. I had to first delete all alarm* files from the watch using the Web IDE, then upload worked fine.

I still have an open update for the GPS recorder. Is there anything I could do that would help you debug it?

gfwilliams, 2020-06-02:
Ok, thanks - that at least really clears it up - it's definitely the `reset` call ending too quickly: https://github.com/espruino/BangleApps/blob/master/js/comms.js#L6

I can't reproduce here (even tried on Windows) so maybe it's something installed on your watch making it take a while to actually reset, that makes things time out. I've just made some changes - please can you try now on the GitHub version?

ps-igel, 2020-06-02:
>please can you try now on the GitHub version

Same error:
`TOAST[error] GPS Recorder update failed, Unexpected response [?7l`

>something installed on your watch making it take a while to actually reset, that makes things time out.

Does the reset load the *boot files? I have one that has code to overwrite the E.showMenu function. Maybe thats slowing it down? Other than that I have no special things installed.

gfwilliams, 2020-06-02:
Sorry, my bad. Was in a hurry and the `git push` failed :)

Glad you got it sorted but I think the underlying issue is now fixed too. Also I'm surprised touchmenu really added that much of a delay - it looks like something that'd execute pretty quick.

Please can you try a full 'hard' refresh of the app loader and try with `touchmenu`? That particular `TOAST[error]` message  should read `<TOAST>[error]`

ps-igel, 2020-06-02:
>I think the underlying issue is now fixed too

Alright, cool! :)

>Please can you try a full 'hard' refresh of the app loader and try

Will do when I have another update to be done and report back.

ps-igel, 2020-06-03:
Ok, so I did not update but upload and saw that the app loader retried after sending reset.
So I would assume the retry cycle also works when updating.

I will still keep an eye on it.

```
<BLE> Sending "\u0003\u0010reset();\n"
puck.js:362 <BLE> Sent
comms.js:8 <COMMS> reset: got ""
comms.js:11 <COMMS> reset: no response. waiting 4...
puck.js:362 <BLE> Busy - adding Puck.write to queue
puck.js:362 <BLE> Executing {"type":"write","data":"\u0003"} from queue
puck.js:362 <BLE> Sending "\u0003"
puck.js:362 <BLE> Sent
comms.js:8 <COMMS> reset: got ""
comms.js:11 <COMMS> reset: no response. waiting 3...
puck.js:362 <BLE> Busy - adding Puck.write to queue
puck.js:362 <BLE> Executing {"type":"write","data":"\u0003"} from queue
puck.js:362 <BLE> Sending "\u0003"
puck.js:362 <BLE> Sent
comms.js:8 <COMMS> reset: got ""
comms.js:11 <COMMS> reset: no response. waiting 2...
puck.js:362 <BLE> Busy - adding Puck.write to queue
puck.js:362 <BLE> Executing {"type":"write","data":"\u0003"} from queue
puck.js:362 <BLE> Sending "\u0003"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Received "\u001b[?7l\r\n ____        "
puck.js:362 <BLE> Received "         _ \r\n|  __|_"
puck.js:362 <BLE> Received "__ ___ ___ _ _|_|___"
puck.js:362 <BLE> Received " ___ \r\n|  __|_ -| . "
puck.js:362 <BLE> Received "|  _| | | |   | . |\r"
puck.js:362 <BLE> Received "\n|____|___|  _|_| |_"
puck.js:362 <BLE> Received "__|_|_|_|___|\r\n     "
puck.js:362 <BLE> Received "    |_| espruino.com"
puck.js:362 <BLE> Received "\r\n 2v05.140 (c) 2019"
puck.js:362 <BLE> Received " G.Williams\r\n\r\nExecu"
puck.js:362 <BLE> Received "tion Interrupted\r\n>"
comms.js:8 <COMMS> reset: got "\u001b[?7l\r\n ____                 _ \r\n|  __|___ ___ ___ _ _|_|___ ___ \r\n|  __|_ -| . |  _| | | |   | . |\r\n|____|___|  _|_| |___|_|_|_|___|\r\n         |_| espruino.com\r\n 2v05.140 (c) 2019 G.Williams\r\n\r\nExecution Interrupted\r\n>"
puck.js:362 <BLE> Sending "\u0010E.showMessage('"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "Uploading\\nverti"
puck.js:362 <BLE> Sent
puck.js:362 <BLE> Sending "calface...')\n"
puck.js:362 
```

closed 2020-06-03 by ps-igel