espruino/BangleApps #464: Alarm clock: Alarm confirmation immediately goes to launcher

nxdefiant, 2020-05-29:
Hi,

when confirming an alarm with "ok" the clock initially goes to default clock and a second later into the launcher.

Expected: The alarm confirmation dialog only returns to the default clock.

gfwilliams, 2020-05-29:
Ahh - are you pressing BTN2 pretty slowly? If you tap it quickly is it fine?

nxdefiant, 2020-05-29:
For my circumstances it is pretty fast, lets say about half a second.

nxdefiant, 2020-07-12:
Same for [Chrono Widget](https://github.com/espruino/BangleApps/blob/master/apps/chronowid/app.js#L14) when I exit the menu it goes first to the clock and a second later into the launcher. When I skip the storage.writeJSON() call it stays in the clock. Something in the storage.writeJSON() call makes it go to the launcher..

m0n5t3r, 2020-10-14:
I think this is fixed in 2v07, I haven't seen the behaviour since upgrading (it definitely happened in 2v05, I don't remember it happening in 2v07 stable and it definitely doesn't happen in my current bleeding edge / 2v07.67)

gfwilliams, 2020-10-15:
Great - yeah, I have a feeling this got fixed too

closed 2020-10-15 by gfwilliams