espruino/BangleApps #465: stopwatch: nondeterministic button 1 behavior

nxdefiant, 2020-05-30:
When pressing button 1 while the display is off the clock sometimes does only turn the lcd on, sometimes also executes a function.

How to reproduce:
1. start the stopwatch
2. press go (button 2)
3. wait for the display to turn off
4. press button 1 (lap) => The display will only turn on.
5. Wait for the display to turn off again
6. press button 1 (lap) => The display will turn on and produce a lap
7. repeat from 3.

I'm not able to figure out when the clock does 4. or 6, it seems to be random to me.

This can be reproduced in some other apps, e.g. scrolling in settings or launcher.

v1nc, 2020-06-30:
I also recognized this, but for me it happend with button 2. Sometimes it only turns on the screen, sometimes it also stops the time.

gfwilliams, 2021-02-02:
Should be fixed now with 2v08 and later firmwares

closed 2021-02-02 by gfwilliams