espruino/BangleApps #485: weather: no wind direction

pasniak, 2020-06-04:


nebbishhacker, 2020-06-05:
I keep meaning to look into this. As far as I can it's an issue on the GadgetBridge end of things.

rigrig, 2020-07-08:
There seem to be issues on both ends :-(
* The BangleJS weather app handles the format as specified on on https://www.espruino.com/Gadgetbridge: 
  > t:"weather", temp,hum,txt,wind,loc - weather report

  `wind` would be wind speed, there is no wind direction field listed, so the app wouldn't handle it anyway
* The GadgetBridge code won't send `windDirection`, only `windSpeed`: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/75a0415f5d26a52fed8712225e5d0ad1b3268ae2/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L503
* I logged `GB` messages, and `wind` is exactly zero, even though the "Weather notification" app on my phone (where GadgetBridge gets these values) says it should be "4 km/h":
`{"t":"weather","temp":288,"hum":94,"txt":"Light rain","wind":0,"loc":"Hengelo"}`

pasniak, 2020-08-12:
Opened issue at https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/1974

rigrig, 2021-02-28:
Gadgetbridge was fixed: https://codeberg.org/Freeyourgadget/Gadgetbridge/commit/8599d2f8dbc6cf65dcb2e14e0ef6dd5da5b25d77   
it now sends wind direction in `wdir` (looks like degrees, not sure how granular though, saw `90` so far), and `wind` contains the correct speed.   
So wind speed now works, need to look into adding direction to the app+widget.

PR #763 (2021-06-21): weather: show wind direction, check bluetooth status when showing "weather unknown"
Also improve Readme a bit

Fixes #485,#762
commit 863923a3 (2021-06-21): fix #485: weather: add wind direction
closed 2021-06-21 by gfwilliams