espruino/BangleApps #498: Animated Clock other Images

v1nc, 2020-06-10:
The Animated clock app is so great! It fits perfectly in BangleJS look.

I wonder if its possible to add other images from http://www.effectgames.com/demos/canvascycle/. I am desperately in love with the jungle waterfall in the morning. 
Could you explain how you ported the data of a specify image?

Best Regards

gfwilliams, 2020-06-11:
Yes, sure! There's some info in https://github.com/espruino/BangleApps/blob/master/apps/animclk/create_images.js

So you need to get the `xyz.LBM.js` file - they are actually linked at https://github.com/jtsiomb/colcycle/wiki if you don't want to dig into the page to get them.

Then modify `create_images.js` to use that file and run it. It'll also write `Cycles` to the console, which you need to copy & paste into the app itself.

You may also want to modify https://github.com/espruino/BangleApps/blob/master/apps/animclk/create_images.js#L32 to adjust the framing of the image (the 640x480 image has to be cropped - I guess you might be able to scale by 2 but the images use dithering that will probably look odd if rescaled).

If you get some nice ones maybe you could attach them here and I could look at adding a menu to the upload so the image can be chosen

rigrig, 2020-07-16:
These look pretty amazing, but I don't think it's actually allowed to distribute these images?
Looks like the rendering code is open source, but the actual artwork is encumbered by copyright and licensing restrictions: http://www.effectgames.com/effect/article-Q_A_with_Mark_J_Ferrari.html#_section_1_4

Having an upload page grab them from that website might be ok maybe?

v1nc, 2020-07-16:
> These look pretty amazing, but I don't think it's actually allowed to distribute these images?
> Looks like the rendering code is open source, but the actual artwork is encumbered by copyright and [...]

Maybe a page to choose the base image and the cutout is more comfortable to choose other images. I think we should be allowes to display a preview of the image that would be shown on the screen, if the user uploaded the image itself.
The question is, is the user allowed to upload these files somewhere?

gfwilliams, 2020-07-16:
Yeah, I figured given the age of these and the fact they were available online (and the entire games they are from are freely downloadable as abandonware) I wasn't likely to have issues with the single image (and would of course remove it at the first complaint), but I think if we started hosting loads then that could be an issue.

I think linking to the `xyz.LBM.js` that were hosted elsewhere would be fine - but I bet there are no CORS headers on that site so we couldn't download directly :(

gfwilliams, 2020-07-16:
Also, I just spotted http://www.effectgames.com/demos/worlds/

While not animated, it'd be pretty fun to have a watch face that changed itself depending on the time of day

rigrig, 2020-07-16:
Some of those actually are animated, and you could even link them to the weather.
"Jungle Waterfall" for example has Clear, Cloudy, Rain, Morning, Afternoon, and Night variants. 

I looks like you can grab them as javascript from http://www.effectgames.com/demos/canvascycle/image.php?file=V29
so I suppose (unsafe!) downloading would be possible using JSONP by providing our own `CanvasCycle.processImage`...

gfwilliams, 2020-07-16:
> I suppose (unsafe!) downloading would be possible using JSONP by providing our own CanvasCycle.processImage

Wow, that's a great idea! 

```
CanvasCycle={processImage:c=>console.log("yay",c)}
var s = document.createElement("script");
s.type = "text/javascript";
s.src = "http://www.effectgames.com/demos/canvascycle/image.php?file=V08AM&callback=CanvasCycle.processImage";
document.body.appendChild(s);
```

As long as we hard-coded the URLs so you couldn't have some random URL in there I think we'd be more or less ok. Given the amount of stuff out there that uses files hosted elsewhere I think we'd be ok doing this in the iframe of the interface.html

v1nc, 2020-07-16:
> Also, I just spotted http://www.effectgames.com/demos/worlds/
> 
> While not animated, it'd be pretty fun to have a watch face that changed itself depending on the time of day

Yes! I also thought about that. I already tried some cutouts with the jungle waterfall and it looks beautiful(image below).

Unfortunately the cycling is somehow wrong for all scenes except night and rain. The windows of the houses flicker white and black. I tried to play with the cyling but could not fix it, maybe someone can look into it. 
[image]

gfwilliams, 2020-07-16:
Did you update the `cycles` with the new one from the file?

Looks like actually there's a `timeline` entry which contains which colour palette *and* cycles to use for each specific time of day, so I guess you'd need to use that

v1nc, 2020-07-16:
Yes, I used the cycles `create_image` printed. Only the window animation is wrong, the rest is fine.

gfwilliams, 2020-07-16:
Probably you'll find some entry in `cycles` that only covers a very small range of numbers (2?) - if you delete it, I bet it'll stop the window.

v1nc, 2020-07-16:
> Probably you'll find some entry in `cycles` that only covers a very small range of numbers (2?) - if you delete it, I bet it'll stop the window

Thanks for the hint, removing the cycle with `low: 232, high: 237` does it.

How do we proceed? Would be pretty neat if we also make the cutout editable in the app loader :smiling_face_with_three_hearts:

gfwilliams, 2020-07-16:
Great! I think it's just a matter of doing something like this: https://www.espruino.com/Bangle.js+Custom

There's an example for `imgclock` as well. I'd reckon you could load the JS file, render the image to a canvas, and then allow someone to drag the clip rect around as well?

rigrig, 2020-07-16:
Maybe also have a look at how http://www.effectgames.com/demos/worlds/palette.js handles cycling.
It looks like the Bangle app doesn't do anything with the `rate` and `reverse` parameters, that might explain why those windows were misbehaving.

gfwilliams, 2020-07-17:
The issue with `rate` is updating the screen takes a bunch of time. It'd be fine if it was 'update X then update Y every 5th frame' but the rates are often things like 1.2x. When I'd experimented I couldn't get them looking nice :(

But yeah, reverse could be useful, and the online version appears to blend the palette colours, and you could try that (although I'm not sure if Espruino is fast enough)

gfwilliams, 2021-02-02:
Closing this due to lack of activity

closed 2021-02-02 by gfwilliams

v1nc, 2021-03-17:
> Closing this due to lack of activity

Sorry for my inactivity. Are you not interested in making the image and cutout selectable or do you wait for me to implement it? 
Looks like there are not many people who need it.
I am fine with my waterfall a the moment, but would be nice to see the selection feature.

Btw thanks again for your great work, I just saw how many new apps were released and am very happy about it. Also bluetooth connection is a lot more stable with the latest firmware :1st_place_medal:

gfwilliams, 2021-03-18:
Hi! No problem - I've just got a lot on at the moment so I've got to pick what I work on.

Personally I think having a selectable cutout might be a step too far given how much the clock face actually get used - but I think it'd be nice to have some other scenes available, just with the areas predefined. Of course if you wanted to make the area configurable too that'd be great.

I could be wrong, but it seems that https://github.com/espruino/BangleApps/issues/498#issuecomment-659385656 no longer works, so that limits the options a bit - I think we'd have to host everything.

With the 'random clock' widget, I guess folks may want to have more than one animated clock on their watch at the same time - so maybe the most sensible all-round option is to just add some whole new clock 'apps' along the lines of this app?