espruino/BangleApps #499: Global style settings

v1nc, 2020-06-12:
It would be pretty neat to have a global style settings for background color, text color and accent color. We could then update all apps to use the colors if they are set. 

This would make the BangleJs experience more consistent. What do you think about it?

gfwilliams, 2020-06-12:
I was wondering about this, but I don't know how many apps would implement it reliably...

What's your main reasoning? You'd like to be able to make things black on white?

v1nc, 2020-06-12:
> I was wondering about this, but I don't know how many apps would implement it reliably...
> 
> What's your main reasoning? You'd like to be able to make things black on white?

I was working on a terminal inspired clock with configurable colors and thought it would be nice to use the config for other apps. 
I know this wont work for all apps because some have custom styling, but most apps have one background color and one text color. 

I am willing to add it to all apps that would work with it.

jameshowe, 2020-06-28:
How about introducing a `7chname.theme.js` file onto the app / widget utility belt? They could be loaded _before_ any of the `app.js` / `widget.js` files which would make them available to use in your app, the Bangle could provide a default theme for a consistent L&F (they obviously don't _have_ to use this, but it's a good starter for 10).

An example `*.theme.js` file might look something like:
```
(() => {
  THEMES['7chname'] = {
    bgColor: '#ff0000' // override the background color only
  };
})();
```
Idea being, if the Bangle provides a default theme, then in your custom theme you only need to override the settings you want to change. Thoughts are Bangle could then pre-configure the Graphics as part of the app loading so us lazy Devs don't need to :)
```
// load theme
const theme = Object.assign({}, THEMES.default, THEMES[app_name]); // presuming the name could be easily pulled from the file / app.json?
g.reset();
g.setColor(theme.color);
g.setBgColor(theme.bgColor);
g.setFont(theme.fontName, theme.fontSize);
...
// load widgets / apps
```
Alternatively, the Bangle can provide a hook that an app / widget can implement to provide it with the theme (which gives you the opportunity to leverage an existing apps theme as they should be loaded at that point).

Just an idea :)

v1nc, 2020-06-29:
> How about introducing a `7chname.theme.js` file onto the app / widget utility belt? They could be loaded _before_ any of the `app.js` / `widget.js` files which would make them available to use in your app, the Bangle could provide a default theme for a consistent L&F (they obviously don't _have_ to use this, but it's a good starter for 10).
> 
> An example `*.theme.js` file might look something like:
> 
> ```
> (() => {
>   THEMES['7chname'] = {
>     bgColor: '#ff0000' // override the background color only
>   };
> })();
> ```
> 
> Idea being, if the Bangle provides a default theme, then in your custom theme you only need to override the settings you want to change. Thoughts are Bangle could then pre-configure the Graphics as part of the app loading so us lazy Devs don't need to :)
> 
> ```
> // load theme
> const theme = Object.assign({}, THEMES.default, THEMES[app_name]); // presuming the name could be easily pulled from the file / app.json?
> g.reset();
> g.setColor(theme.color);
> g.setBgColor(theme.bgColor);
> g.setFont(theme.fontName, theme.fontSize);
> ...
> // load widgets / apps
> ```
> 
> This would work well for apps / widgets just loading their own theme, but gives you the flexibility of leveraging other apps themes as well (you would just need to manually override the Graphics settings).
> 
> Just an idea :)

This sounds great. Apps that currently dont care about graphics would work out of the box with the default theme, apps that use custom graphics would still look the same.

I only wonder about how to let the user edit the global graphic settings. I can not imagine a comfortable way to input colors on the BangleJs. Maybe we have to manage this via the App loader.

If there is work I can take over for you, feel free to explain what :)

gfwilliams, 2020-06-29:
I think the issue with trying to override graphics is pretty much everything uses `setColor` at some point. I think this'll have to be something that has to be implemented in every app.

I'm not sure about the automatic theme.js file purely because right now app load times can be a bit high if you have a bunch of files on your watch - another file search/access (or two) would make that quite a bit worse (also getting the app ID is actually not possible at the moment). But the idea of per-app themes and overriding sounds pretty neat.

I guess asking app developers to just add this code at startup when they wanted a theme wouldn't be a big deal?

```
// normal theme
var THEME = Object.assign({color:-1,bgColor:0}, require("Storage").readJSON("theme.json",1));
// or allow per-app theme too
var THEME = Object.assign({color:-1,bgColor:0}, require("Storage").readJSON("theme.json",1), require("Storage").readJSON("myapp.theme.json",1));
```

And things like `E.showMenu` could then check for the presence of the THEME variable too.

To choose the themes on the Bangle we could just have a palette of pre-set colours to choose from. And obviously one or two global themes pre-made.

jameshowe, 2020-06-29:
> I think the issue with trying to override graphics is pretty much everything uses setColor at some point.

Agree, having thought more about it, it wouldn't be of much value to pre-configure the graphics, you'll more than likely be changing this as you render (e.g. borders, backgrounds, text etc.)

> I'm not sure about the automatic theme.js file purely because right now app load times can be a bit high if you have a bunch of files on your watch - another file search/access (or two) would make that quite a bit worse

Yeah fair point, plus I think JSON files make more sense, I only thought JS as I was trying to align with how the current eco-system works (ala widget loading). One point though - does making it an "opt-in" approach not just solve the problem temporarily? i.e. if all apps eventually come to support themes, does the load time problem not just manifest itself in a different way?

I think based on what you've said, I'm back to wondering whether there is much benefit to this now 🤔 In the short-term, probably not, longer term though I think we could introduce the notion of a UI library that would make rendering a whole lot simpler for users which is where the real power of themes would come in :)

v1nc, 2020-06-30:
> Agree, having thought more about it, it wouldn't be of much value to pre-configure the graphics, you'll more than likely be changing this as you render (e.g. borders, backgrounds, text etc.)
>
>
> Yeah [...]

I agree with most of that. Idk about plans about an UI library, but this would be very powerful.

Anyway I see some benefit for a global style setting in the short-term. We(or I 😬) can edit all current apps that do not use a specific color sheme, so they use the global setting for background and text color. This would make BanlgeJS experience a lot more consistent and configurable, while not influencing app loading time.

jameshowe, 2020-06-30:
> Idk about plans about an UI library, but this would be very powerful.

I'd be happy to start one :) @gfwilliams I presume if I do this right, you could re-use it for the Pixel too? (it would just wrap up the [Graphics](https://www.espruino.com/Reference#Graphics) class).

> We(or I 😬) can edit all current apps that do not use a specific color sheme, so they use the global setting for background and text color

Sounds good - I guess we just need to agree the approach. @gfwilliams how about this:

- As per your suggestion, we have a single `theme.json` file that all apps should check for and load if it exists, it's then up to them to adhere to using this throughout
- To _create_ that `theme.json`, how about we have a system folder `themes` on the Bangle and we can drop a bunch of JSON files in there that the Bangle can enumerate & list via Settings? This would make it easy for people to contribute e.g.
```
theme.json
- themes
-- stylea.json
-- styleb.json
```
- On selecting a theme, the JSON file simply creates or replaces the existing `theme.json`

gfwilliams, 2020-06-30:
There is actually already a pretty full featured UI lib at https://www.espruino.com/ui - just nobody seems to be using it :)

Having just a global theme seems like the best bet - if done the same way for each app it should be trivial to extend to pre-app settings at some point later if needed.

I'm afraid there aren't folders. I think for now it's better to just keep the themes listed as a blob of JSON in the Theme settings app (it's only going to be a handful of colours per theme). PRs for that should still be pretty trivial, and it can always be changed at a later date.

closed 2021-05-26 by gfwilliams