espruino/BangleApps #506: Alarms: unable to save

pasniak, 2020-06-15:
I tried to schedule New Alarm. I selected hour, minute, then Save and pressed BTN2 and... nothing happened! I first thought it's normal behaviour but the watch did not vibrate at all nor any message was displayed.

nxdefiant, 2020-06-15:
Could it be the problem with the multiple alarm.json?

gfwilliams, 2020-06-22:
Hi - this was fixed a month or two ago, but you'll need to update your firmware: https://www.espruino.com/Bangle.js#firmware-updates

Please can you ensure you're on up to date watch firmware? 2v06 at the moment.

Then do `About -> Install default apps` from the app loader to clear storage - and the issue should go away and no longer appear.

pasniak, 2020-06-22:
About (btw: which unfortunately comes alphabetically before Alarms...) confirmes I have 2.06 (just flashed it). I installed default apps (so only have 5 items in BTN2 menu). Still `Save` does nothing.

gfwilliams, 2020-06-22:
Can you try `Remove all apps` first? The `Install Default Apps` button should have done it, but maybe it failed somehow.

pasniak, 2020-06-22:
I just did that full cleanup (so no boot) and install /9 default apps. I do `Alarms` > `New Alarm` > `Save`. Nothing else - and nothing worthwhile happens... is alarm supposed to appear in some list?

gfwilliams, 2020-06-23:
Thanks - just tried, and I can reproduce here.

Sorry about this - it looks like the change to add 'auto-snooze' was completely untested :( 89909c07464458e665260c4251348c625fa2d5ee

commit 83098049 (2020-06-23): 0.10: Fix auto-snooze option (this stopped new alarms being added) (fix #506)
closed 2020-06-23 by gfwilliams