espruino/BangleApps #509: GATT Failed while upgrading firmware

pasniak, 2020-06-22:
In attempt to fix #506 I am trying to upgrade from 2.05 to 2.06. I downloaded espruino_2v06_banglejs.zip.
On watch: DFU Start
On Nordic app: after pressing Upload I get Uploading...
After a while on the tablet: GATT Failed

pasniak, 2020-06-22:
nevermind, I was not selecting DfuTarg. You could add this: `If you see **GATT Failed** message make sure you selected DfuTarg NOY banglle.js as device`

closed 2020-06-22 by pasniak

gfwilliams, 2020-06-22:
Thanks. Actually I just had another report about this as well. You followed the instructions here?

https://www.espruino.com/Bangle.js#firmware-updates

I think realistically some photos/videos would help.

pasniak, 2020-06-22:
Too late it worked for me - maybe next time.

pasniak, 2020-06-23:
2.06 seems less stable as far as GPS (I get "Searching for GPS time" followed by black screen + no button response).  Are there any release notes?

gfwilliams, 2020-06-23:
http://www.espruino.com/ChangeLog and http://forum.espruino.com/conversations/349172/newest/ ?

Worth making sure all your JS firmware is up to date first...

pasniak, 2020-06-23:
[image] is that the symptom? Could you please point me to the doc? (and maybe add link under firmware install?)

pasniak, 2020-06-23:
All apps were reinstalled yesterday as I flashed Espruino v2.06. In About:
Version 2v06
Commit 21168a2b
Bootloader v0.19
Launcher v0.03
Settings v0.20

gfwilliams, 2020-06-23:
Looks like you have the numerals clock installed? If you just install another clock it'll be fine.

Try installing default apps and work from there.

Honestly, looking at the code of numerals clock with the error, it's not nice at all. While it seems it should work, it just feels like it'd break at some point. I'll change it and it should update.

pasniak, 2020-06-23:
That worked. You could include version on which apps were tested, like Android does.

gfwilliams, 2020-06-24:
> You could include version on which apps were tested, like Android does.

Not sure I understand... 'Install default apps' should be the set of apps that are tested for each release?

Unfortunately it's not really feasible to test every app for each release unless I can come up with some magic Bangle.js + webcam thing.

pasniak, 2020-06-24:
Couldn't apps be automatically tested on the emulator? The clocks at least.