espruino/BangleApps #515: New app: GPS settings

gfwilliams, 2020-06-27:
The UBlox GPS is actually pretty configurable. There is an example of sending commands to it in the 'testing' folder but we should expand this and make it into an app - allow:

* Low power mode
* Set GPS time from clock
* Choose GPS/GLONASS/Beidu
* AGPS
* what GPS data gets reported
* GPS report speed
* Anything else from the datasheet?

jl-cs, 2020-07-03:
What about [Galileo](https://en.wikipedia.org/wiki/Galileo_(satellite_navigation))?

gfwilliams, 2020-07-03:
If the GPS receiver does it (which looks promising), it could be enabled: https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29_Public.pdf 

However it might be on by default

gfwilliams, 2021-04-14:
Fixed with @hughbarney's great work

closed 2021-04-14 by gfwilliams