espruino/BangleApps #523: GPS Recorder does not log timestamps

4thex, 2020-07-08:
The current format of the KML file currently something looks like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2">
  <Document>
    <Placemark>
      <name>Bangle.js Track 0</name>
      <LineString>
        <coordinates>
          -78.77696,36.92808,137.6
          -78.77696,36.92808,137.6
          -78.77696,36.92808,137.6
        </coordinates>
      </LineString>
    </Placemark>
  </Document>
</kml>
```
I am proposing that the timestamp feature is added so it is evident in the KML file when the GPS observation was made.
The expected KML file would look something like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2">
  <Document>
    <Placemark>
      <name>Bangle.js Track 0:1</name>
      <LineString>
        <timestamp>2020-07-08T19:43:00.000Z</timestamp>
        <coordinates>
          -78.77726,35.92763,154.5
          -78.77719,35.92759,137.1
        </coordinates>
      </LineString>
    </Placemark>
    <Placemark>
      <name>Bangle.js Track 0:2</name>
      <LineString>
        <timestamp>2020-07-08T19:44:00.000Z</timestamp>
        <coordinates>
          -78.77719,35.92759,137.1
          -78.77709,35.92750,120.0
        </coordinates>
      </LineString>
    </Placemark>
  </Document>
</kml>
```
So basically each placemark becomes a line segment with a timestamp. The timestamp should correspond to the starting coordinates. There should also be a placemark at the end of the file so the timestamp for the final observation is also included.

gfwilliams, 2020-07-09:
Sounds like a good idea adding timestamps - time data is stored for each reading so this would be easy to extract.

However you'd end up with hundreds of placemarks, which I think isn't what most people would want, and I'm not sure it would integrate well with most software (do you have anything in mind that you want to import this into?). We could add it as an option but IMO we should have the old one there with just the initial timestamp by default.

4thex, 2020-07-09:
Google Earth does support the timestamps. I also wanted it for my own daily run tracking, so I can see how fast I ran in individual sections, possibly correlating it with heart rate recording. The timestamps could be an option in the settings for the app.

4thex, 2020-07-09:
If I get some time this week, I can work on it and send you a pull request. You can assign the issue to me if you like.

gfwilliams, 2020-07-09:
Actually have you considered just exporting in GPX format? That already marks the time of every GPS coordinate, and there's even provision for including heart rate/etc built into the spec - it looks like a *far* better solution than creating 1000-odd placemarks.

4thex, 2020-07-09:
Thank you! I just verified that, and it works perfectly. I will close the issue.

closed 2020-07-09 by 4thex