espruino/BangleApps #527: Notify API improvements

rigrig, 2020-07-14:
I'm wondering if it might be a good idea to change the notify API a bit:
Turn `options.size` into a "suggested size", and pass the actual available area to `render`.

That way `render` could use e.g. a larger font for fullscreen notifications.

My idea would be that
- `area` starts out as the whole notification
- drawing `title`,`src` and/or `icon` shrinks `area`
- `body` text (if present) is drawn inside `area`
- `render` (if present) is called with `area`

So you *could* pass both `body` and `render` to e.g. draw a border around the text, but if you want to get fancy you need to render it yourself.

Or a simpler alternative: just an optional additional boolean argument passed to `render`, which indicates it *could* ignore the `y` parameter and instead draw to the whole screen.

gfwilliams, 2020-07-15:
For others - current docs at: https://github.com/espruino/BangleApps/tree/master/apps/notify

This sounds like a great idea! I'd definitely be up for that. 

I think having `y` is handy as the non-fullscreen notifications position themselves at particularly odd locations - maybe we should have: 

* `y` -> as you now suggest
* `height` -> actual area left to render
* `top` -> the top of the notification - sometimes you may want to draw the notification as normal, but then add something extra on top of it?

rigrig, 2020-07-15:
Yes, you definitely want the expected location, not figure out where to draw yourself. (Making the render function responsible for centering inside the actual available area seems reasonable though)
`area` would be the rectangle inside which you are allowed to draw, probably one of

1. `[x1,y1,x2,y2]`: same as `g.draw*` functions, but awkward/uninformative to actually use:
```js
render: area => {
  g.drawRect(area[0],area[1], area[2],area[3]);
  // or more likely, you use this boilerplate in every render function:
  var x1 = area[0], y2 = area[1], x2 = area[2], y2 = area[3];
  g.drawRect(x1,y1, x2,y2);
}
```

2. `{x1,y1,x2,y2}`: similar to `g.draw*` functions, but with clearer naming
```js
render: area => {
  g.drawRect(area.x1,area.y1, area.x2,area.y2);
  // need to calculate size yourself though
  var w = area.x2-area.x1+1, h = area.y2-area.y1+1
}
```

3. `{x,y,w,h}`: seems cleaner, but I suspect most of the time you will be calculating x2/y2 anyway
```js
render: area => {
  g.drawRect(area.x,area.y, area.x+area.w-1,area.y+area.h-1);
  // or
  var x1 = area.x, y1 = area.y, x2 = area.x+area.w-1, y = area.y+area.h-1
  g.drawRect(x1,y1, x2,y2);
}
```

Option 2 looks like the best solution to me, but I'll try how well it works when I change some of the actual code to work with it.

Note that `x` can be >0 if you request an icon:
```js
notify.show({title:"Title", body:"some text", icon:badlyDrawnSmileyFace, 
             render:a=>{g.setColor(0xF800).drawRect(a.x1,a.y1, a.x2,a.y2)}):
```
Draws something like this:
[image]

> sometimes you may want to draw the notification as normal, but then add something extra on top of it?

I don't think there would be any good reason to draw outside the `area`:
Either you don't request a title/icon, so you get a larger `area` to draw. Or you make the notify library responsible for drawing them, but then you can't make any assumptions about where the notification library draws things.
The only exception would be the body text, which you *can* draw on top of.

(Slightly related: Is there some page I overlooked which explains how the display/`LCDOffset` works? I had a quick look at the small notification code and it wasn't really clear to me how it actually works (and where it gets `320`). If not I guess I'll have a look at the datasheet.)

gfwilliams, 2020-07-16:
Sounds good - I think I would have a minor preference for `{x,y,w,h}` over `{x1,y1,x2,y2}` because when drawing widgets I pass in `{x,y}` but I'm not that bordered really.

> Is there some page I overlooked which explains how the display/LCDOffset works?

Not as far as I remember. But basically the display has a 240x320 buffer, while the LCD is 240x240. So when drawing notifications, we write the notification into that 'spare' 240x80px, then shift the LCD rendering area up. The app can continue drawing as normal, but we still display the notification.

So, 320 comes from the buffer height - we render into the bottom bit, but the buffer 'wraps', so when we shift the display up, we get rows 241..319, *then* 0..159

rigrig, 2020-07-16:
> basically the display has a 240x320 buffer

Doh, now it makes sense. I figured it worked like that, but just assumed the buffer held another 240x240 offscreen and I was missing some step how to go from 480 to 320. Thanks!

rigrig, 2020-07-19:
I went with `{x,y,w,h}` after all, it felt cleaner

PR #530 (2020-07-20): Notify api
Fixes #527 

(Having so much duplicate code in both `notify` and `notifyjs` really bugs me, but I guess it can't be helped.)
closed 2020-07-20 by gfwilliams