espruino/BangleApps #533: Search Engine Performance

gfwilliams, 2020-07-21:
How do we let Google index BangleApps properly so a search for "Bangle.js BLE Scanner" goes directly to the "BLE Scanner" app? The URL would be `banglejs.com/apps#blescan`

I made some changes recently to try and help with this: 8beb44510b8d8ca24dc86901023509b754050db8

But there might be more we could do... It's got to help

jjok, 2020-08-08:
I was going to say you could try generating an XML site map, but I'm not sure it considers `#` anchors to be different pages.

Is it possible to make the URL `banglejs.com/apps/blescan` instead? That way it looks like its it's own page. You can do that kind thing with the [History API](https://developer.mozilla.org/en-US/docs/Web/API/History_API).

It would probably help if the apps actually part of initial HTML of the page, rather than added in with JavaScript. How about pre-rendering the apps HTML, so it's already there before the JS kicks in to make all the filtering work? You could do it as part of the build/deployment process. Maybe.

gfwilliams, 2020-08-17:
I guess to make `banglejs.com/apps/blesca` work I'd have to modify the webserver too. Nothing stops me from using query strings though - like `banglejs.com/apps?a=blesca`?

In terms of pre-rendering, I think we're ok as-is. If you google `bangleapps` you can see Google's managed to pull in the content from apps.json by running the JS, so it's just ensuring it knows how to link everything.