espruino/BangleApps #534: Some settings "wake on" for LCD doesn't work

Sblash, 2020-07-21:
Settings like "Wake on FaceUp" and "Wake on Twist" doen't seem to work
Anyone else have this issue?

Sblash, 2020-07-21:
Ok, i've tested the accelerometer with this code

```
Bangle.on('accel', function(acc) {
  console.log(acc);
}); 
```

it seems stucked...or maybe broken?

Didgeridoohan, 2020-07-21:
> Settings like "Wake on FaceUp" and "Wake on Twist" doen't seem to work
> Anyone else have this issue?

Both work fine here. Have you tried a reset? Made sure you're on he latest firmware?

Sblash, 2020-07-21:
> > Settings like "Wake on FaceUp" and "Wake on Twist" doen't seem to work
> > Anyone else have this issue?
> 
> Both work fine here. Have you tried a reset? Made sure you're on he latest firmware?
> 
> Or maybe your accelerometer is faulty.

I've tried firmware version 2v05 and 2v06, nothing changed

gfwilliams, 2020-07-21:
Hi - that's a strange one. It looks like somehow the accelerometer has got misconfigured. I've never seen this before, but the accelerometer is 'always on', so if it did get into a strange mode then resetting Bangle.js wouldn't fix it (only waiting for the battery to fully run down and recharging it would).

Please could you run:

```
Bangle.accelRd(0x0F)    
```

in the console? It should respond `21` if all is going well.

You could also try running:

```
Bangle.accelWr(0x19,0x80)
// followed by:
reset()
```

And see if that helps?

Sblash, 2020-07-21:
> Hi - that's a strange one. It looks like somehow the accelerometer has got misconfigured. I've never seen this before, but the accelerometer is 'always on', so if it did get into a strange mode then [...]

Hi, thank you for the answer!

Bangle.accelRd(0x0F) this return 255
Bangle.accelWr(0x19,0x80) and this return undefined

Sblash, 2020-07-21:
Ok, i tried to let battery completely discharge and then recharged and seems working now
I ran this command
`Bangle.accelRd(0x0F)` 
and it returned 21

Problem fixed, thank you all!

closed 2020-07-21 by Sblash

gfwilliams, 2020-07-22:
That's great! Glad you got it fixed. And now the 'Wake on...' works?

I've never come across that before at all - the accelerometer must have locked up somehow. `Bangle.accelRd(0x0F)` should always return 21 regardless of what happens so it's a very strange one

Sblash, 2020-07-23:
> That's great! Glad you got it fixed. And now the 'Wake on...' works?
> 
> I've never come across that before at all - the accelerometer must have locked up somehow. `Bangle.accelRd(0x0F)` should always return 21 regardless of what happens so it's a very strange one

Yes, it's all working now