espruino/BangleApps #537: ActivePedometerWidget not working with Firmware 2v05

kuleszdl, 2020-07-29:
I was always wondering what was wrong, but the ActivePedometerWidget never worked for me on Firmware 2v05. After upgrading to 2v06, it works now.

I am not 100% sure, but in case this is valid - could it be possible to "flag" apps that are not compatible with the current firmware in the store so people don't get confused?

gfwilliams, 2020-07-30:
Ahh, interesting. When it's an app (eg the Animated Image clock) I try and make it report that you need to update the firmware when it's installed, but widgets can't easily do that.

Yes, I guess we should check firmware versions...

m0n5t3r, 2020-10-14:
it always "worked" for me on 2v05 (as in, counted "steps" - I'm using simple clock), but 
* generated totally bogus results (6km walk? 2k-4k steps; milling around the house / typing for 7-8h? 12k steps; using a hand plane for 30 minutes? 5k steps) 
* caused the watch to be unresponsive pretty often (would not wake on wrist raise / button press for 5-6 seconds)

I ended up uninstalling it; the other pedometer widget is just as useless, so I think the hardware thing that computes / reports steps might be pretty bad at this (and most likely figuring it out in software / with tensorflow from accelerometer data would make the battery last a few hours)

gfwilliams, 2020-10-14:
You're saying you're still running it on 2v05, or on an up to date firmware now?

gfwilliams, 2020-10-15:
Ok, sorry, I misunderstood. I think this can be closed then - very few people should still be using 2v05 now

I'll close this then - there's an issue you spotted open for the accuracy here: https://github.com/espruino/BangleApps/issues/555

Specifically an open Espruino issue at https://github.com/espruino/Espruino/issues/1846 for some more advanced step counting firmware

closed 2020-10-15 by gfwilliams