espruino/BangleApps #539: [Feature request] Make step count data (active pedometer) fetchable via gadgetbridge

kuleszdl, 2020-07-30:
Gadgetbridge supports fetching of data about the steps made from other Gadget already. To make it possible for the bangle.js, I assume the following things would need to be done to accomplish this:

- [ ] store step data history on the bangle.js (not just for the last 24 hours)
- [ ] implement the interface in the gadgetbridge app on the bangle.js to make this data retrievable
- [ ] tell the Gadgetbridge Android app that the bangle.js supports fetching step counter data as well
- [ ] implement a function that erases the step data history once they have been fetched

gfwilliams, 2020-07-30:
Gadgetbridge has a database thing built into it and you need to make a schema per device.

It's not something I had time to dig into when doing the port for Gadgetbridge but I'd welcome someone else looking into it. Once that is done, it should be pretty straightforward to fill in the blanks in the Gadgetbridge Bangle.js handler to receive heart rate/etc data when it's pushed by the Gadgetbridge widget (like we do for battery level at the moment).

Actually catching up after a period of disconnection would be something else entirely I think, but I feel like it's more important to get the basics sorted out first.

gfwilliams, 2021-02-02:
Step count and HRM data is now integrated with Gadgetbridge.

However right now it's live data (not historical) so you need to stay connected to Gadgetbridge for it to work. I think in most cases this is what happens though.

Closing this for now - if fetching historical data into Gadgetbridge becomes a problem we can always look into it

closed 2021-02-02 by gfwilliams

kuleszdl, 2021-05-04:
Thank you, @gfwilliams! However, tbh, I have often the scenario where my Banglejs is not connected to Gadgetbridge, so having the historical data as well would be great. Shall we open a new issue for this?

gfwilliams, 2021-05-04:
Yes, that's probably best - and then reference this one