espruino/BangleApps #540: [Feature request] for the Torch app

PHDPeter, 2020-08-04:
Would it be possible to add in the setting a default colour for the Torch light?
I am thinking just Red (for low light), White (for reading) and Green (the eyes most sensitive colour).
I have only just started with the bangle but would be happy to do this my self if the original make of the app would be ok with that? I mean I would be stealing all there very well made code after all.

gfwilliams, 2020-08-05:
Hi! That sounds like a good idea... I reckon it might be good just to have a settings page added for it? https://github.com/espruino/BangleApps#adding-configuration-to-the-settings-menu

I'm the author and 100% fine with changes :)

Ronin0000, 2022-02-03:
I added a pull request for this, so this can be closed!

closed 2022-02-03 by gfwilliams