espruino/BangleApps #541: Weather App always shows the same icon

phiL0co, 2020-08-06:
My weather app always shows the icon with five horizontal lines (I guess fog?).
It never changes no matter what the weather notification app says.
Same with the widget icon.
Temperature shows the correct value though.

I followed the instructions how to get the weather app to work with gadgetbridge. 

Can anybody help?

Thx!

phiL0co, 2020-08-06:
I suspect it has to do with the chooseIcon function in lib.js.
If none of the keywords is found, the icon for mist is drawn, if I'm correct.

So it seems the correct keywords aren't transferred.

Can the problem be, that the weather notification on my phone is in German?
Or are the keywords nonetheless transferred on English in the background?

Unfortunately I can't find a way to switch languages in the weather notification app to test this...

gfwilliams, 2020-08-06:
That's interesting - it could well be a language issue.

Please could you connect with the Web IDE and then run `require("Storage").readJSON('weather.json')`? That might show what's been received.

phiL0co, 2020-08-06:
[image]

Thx for reaching out so quickly! 
Here's the output!

phiL0co, 2020-08-06:
It's definitely the suspected translation problem!

I added German conditions in lowercase in lib.js and now it's working as intended!

gfwilliams, 2020-08-07:
That's great! I wonder how sensible it is to have translations of weather for every language in lib.js though. I guess maybe it could be configurable on upload?

Or I guess when you set up the weather app, if English is chosen as the language it'd work? But obviously it's not ideal.

phiL0co, 2020-08-07:
Choosing your language (and thus a corresponding lib.js) would be the best option I think to prevent lib.js from getting to big. But I'm new to Bangle.js and programming! ;) 

But I'd happily volunteer to provide the German conditions or maintain the German lib.js!

rigrig, 2021-06-21:
I think adding a way to customize the code on upload is the right way to go.
Also because while [`chooseIcon`](https://github.com/espruino/BangleApps/blob/cd329f5f2032c60b7dc05e118188ad0de0ddbbc1/apps/weather/lib.js#L211-L229) tries to be somewhat flexible, it will probably turn out to be quite tightly coupled to the OpenWeatherMap API messages if anybody ever tries another weather provider.

I haven't looked at custom apps too closely yet, but it would hopefully be doable to just have people select a weather provider, and then concatenate the appropriate `chooseIcon` code?

An alternative solution might be to make Gadgetbridge include the [Weather condition code](https://openweathermap.org/weather-conditions#Weather-Condition-Codes-2), but the idea of coupling that tightly to a proprietary API doesn't really appeal to me.

nebbishhacker, 2021-09-15:
Having Gadgetbridge include the weather condition code or similar sounds ideal to me. IIRC, GadgetBridge already includes some logic to map between other weather provider's APIs and OpenWeatherMap condition codes, so I don't think the coupling is likely to be an issue in practice.

myxor, 2022-01-08:
This should be fixed via https://github.com/espruino/BangleApps/pull/1205
See issue https://github.com/espruino/BangleApps/issues/1049

Ronin0000, 2022-02-18:
This should be closed now that it is fixed.

closed 2022-02-21 by gfwilliams