espruino/BangleApps #544: Global scheduled silent/dark mode

rigrig, 2020-08-12:
As mentioned in https://github.com/espruino/BangleApps/issues/239#issuecomment-610819750 and https://github.com/espruino/BangleApps/issues/239#issuecomment-610848467

It would be nice to have a global silent mode.

Basic functionality:
* During silent mode: disable buzzer+beep, screen only turns on for BTN press
* Settings for automatic start and stop time (both also with option to set no time)
* Manual enable/disable

Nice to have:
* Some way for apps to override it, e.g. the alarm app.
* API for apps to enable/disable it: So you could for example write an app that sets sleep mode for a specific gadgetbridge notification, without messing with the silent mode code.

rigrig, 2021-03-17:
Until now I was thinking about a boot app that would override things like `Bangle.buzz()`, `Bangle.beep()` etc., and provide something like `Bangle._buzz` for apps like the alarm that *really* want to buzz. That seems messy though, and probably would break (new) apps unexpectedly.

I think a better way to go about this would be to add a global `quiet` setting, and change widgets to respect it:
  - `quiet = 0`: normal
  - `quiet = 1`: priority only (e.g. alarm)
Not too sure about also adding:
  - `quiet = 2`: complete silence

I'm thinking we only implement it for widgets (and clock faces, if any of them beep), on the assumption that if you actively opened an app, there is no need for quiet mode.

This means that most widgets could just do something like
```js
if (!(require('Storage').readJSON('setting.json',1)||{}).quiet) {
    Bangle.buzz()
}
```
Then users can always manually adjust the setting, the fallback is normal behavior, and we can have separate apps that manage the schedule by simply changing that setting. (I'm thinking a: simple daily on/off schedule managed through the watch, and b: complicated schedule managed with `custom.html`)

gfwilliams, 2021-03-18:
Thanks - yes, that sounds like a pretty good solution. It also allows apps to maybe decide to do something different in the case where buzz/beep is actually disabled

PHDPeter, 2021-03-23:
Would it be worth having a similar thing for LCD time off/ LCD brightness?
It should be the same process but I get it if people think it is too much.
Would this be something that will need the cutting edge firmware or can I start adding it to apps now? (I mean if I have the quiet value added to setting.json myself.

rigrig, 2021-03-23:
> Would it be worth having a similar thing for LCD time off/ LCD brightness?

Ooh, definitely I think. Apart from the timeout/brightness, disabling wake on movement/touch would be nice.

I'm working on a pull request here: https://github.com/espruino/BangleApps/pull/697
Right now it adds a manual setting, and apps check the setting before buzzing/beeping/turning the screen on. I'll see about adding some quiet mode LCD settings.

rigrig, 2021-03-25:
#697 is nearly finished, but...    
I'm not too sure "Quiet Mode" is the best name, maybe "DnD" (or even something else) would be better?    
I managed to cram "Quiet Mode" in most menus, so "DnD" will definitely fit (and possibly even "Do not Disturb" in one or two places) and probably look a bit nicer.   
I *think* I picked "Quiet Mode" over "DnD" because I figured it was more descriptive as setting. But that was before it got a submenu, so now we can show the full name in there (and it's probably a fairly well-known abbreviation anyway)

gfwilliams, 2021-03-26:
I think `Quiet Mode` is fine - as you say, 'Do not Disturb' might not fit in some menus and 'DnD' isn't very helpful.

rigrig, 2021-12-02:
Closing this issue, as #697 was merged months ago, adding the [Quiet Mode Schedule and Widget](https://banglejs.com/apps/#qmsched) app.

closed 2021-12-02 by rigrig