espruino/BangleApps #547: What is red "CALLBACK" message above lower edge?

phiL0co, 2020-08-14:
Can anybody tell me what a red message saying CALLBACK means?
It appears under the watchface. 
I noticed that for instance the chrono widget timer stops if this message appears.
The message disappears after refreshing the screen.

gfwilliams, 2020-08-17:
It's a warning from the watch itself to say that the JavaScript app/widget code caused an error that wasn't handled.

If you could find a way to reliably reproduce this (and what widgets are installed/app running at the time) then we could probably track down what the problem is - but it's basically impossible to track down without that.

phiL0co, 2020-08-17:
OK, Thx Gordon!
I think it has to do with gadgetbridge disconnects.
I'll try to narrow it down!

Didgeridoohan, 2020-08-17:
I see this message one in a while as well... Will try to get some troubleshooting in to narrow it down.

kuleszdl, 2021-03-21:
I have seen this message as well. I am running with firmware 2v08 and the following apps:

- About
- Active Pedometer
- Alarms
- App Manager
- BangleRun
- BatteryWidget
- Bootloader
- Gadgetbridge
- Launcher
- Morph Clock+
- Notifications
- Welcome

@phiL0co @Didgeridoohan Maybe you could compare this list of apps to yours so we could narrow down the list of apps that might be causing this?

gfwilliams, 2021-03-22:
Thanks - if the error happens over a wide variety of apps it might be worth checking Widgets first - since they will run in the background for a bunch of apps, so one widget might cause the message to appear in many different apps.

closed 2021-12-02 by gfwilliams